olikraus/u8g2 #2423: NodeMCU ESP8266 display not working

davidvilanova, 2024-04-11:
Hello,
I have a NodeMCU ESP8266 display . I have tried to run a hello world script to display but it´s not doing anuthing

This is the log:
```
Variables and constants in RAM (global, static), used 29040 / 80192 bytes (36%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ DATA     1504     initialized variables
╠══ RODATA   1224     constants       
╚══ BSS      26312    zeroed variables
. Instruction RAM (IRAM_ATTR, ICACHE_RAM_ATTR), used 61355 / 65536 bytes (93%)
║   SEGMENT  BYTES    DESCRIPTION
╠══ ICACHE   32768    reserved space for flash instruction cache
╚══ IRAM     28587    code in IRAM    
. Code in flash (default, ICACHE_FLASH_ATTR), used 247552 / 1048576 bytes (23%)
║   SEGMENT  BYTES    DESCRIPTION
╚══ IROM     247552   code in flash   
esptool.py v3.0
Serial port /dev/cu.usbserial-110
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 68:c6:3a:fc:0c:4d
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 283024 bytes to 207659...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 283024 bytes (207659 compressed) at 0x00000000 in 20.1 seconds (effective 112.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

This is the script I´m trying:

```
#include <U8g2lib.h>

// Initialize U8G2 for SSD1306 128x64 display using software I2C
// Parameters: (rotation, clock, data, reset)
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 12, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.drawStr(0,20,"Hello World!");
  u8g2.sendBuffer();
}

```

 From the specification :
SDA (D6 / GPIO12) and SCL (D5 / GPIO14)

I´m a beginner so don´t know what is wrong ?? Nothing is displayed. I have tried with a blink led example and it works

davidvilanova, 2024-04-11:
Ok got it.

Had to invert clock and data values as follows!!!!  It works now !!!

`U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ 12, /* data=*/ 14, /* reset=*/ U8X8_PIN_NONE);
`

KlimDos, 2024-04-13:
two days... 2 days spend with it.. and. it just coz on their darn pdf it says 14 12 ...

thank you @davidvilanova

dynaram, 2024-05-25:
Thank you so much I have been having the same issue. Is finally working for me thanks to your fix comment. Now lets learn some more. Newbie here.