olikraus/u8g2 #2432: Can I use u8g2 with SSD1357 mono mode?

stayfooll, 2024-04-21:
Can I use u8g2 with SSD1357 mono mode?

I have a project using an SSD1357 WINSTAR 128x64 OLED Model No: WEP012864UWPP3D00000.

The datasheet mentions mono mode.

Is it possible for me to use u8g2 with the SSD1357 mono mode?

If yes, how can I modify the existing code to support SSD1357 mono mode?

Should I start with the function u8x8_cad_SendData?

Thanks.

[image]

SSD1357 WINSTAR 128x64 OLED datasheet:
[https://drive.google.com/file/d/1YEhlR5Wt07mSpj_zeLDudUAPKW4Alpdt/view?usp=drive_link](url)

olikraus, 2024-04-27:
it is not supported...

stayfooll, 2024-04-29:
Got it. Thank you!

stayfooll, 2024-04-29:
Close the issue.

closed 2024-04-29 by stayfooll