olikraus/u8g2 #2433: SH1107 128x128 OLED text jumbled around

chauens, 2024-04-22:
### Discussed in https://github.com/olikraus/u8g2/discussions/2430

<div type='discussions-op-text'>

<sup>Originally posted by **chauens** April 18, 2024</sup>
Hi,
I am using the Adafruit 5297 128x128 OLED with the U8g2 library and have a problem with the way the text is displayed. The OLED is  mounted into an enclosure with its top pointing up as described on Adafruits website. Using a simple code to display "Hello World!" starting at line position 0 shows: "World!        Hello" .
If I use rotation 90 degrees the text shows correct, correct sequence, but of course turned sideways... 90 degrees.
I can't turn the OLED in the enclosure and I want to use U8G2 since it offers the power save mode and more importantly has the option to define the boards I2C SDA/SCL pins since my board (ESP32 Feather) is not using the defaults.
Any ideas how to get around this? Any help appreciated.

This is the test code:
#include <U8g2lib.h>
#include <Wire.h>
#define I2C_Adr 0x3d  //OLED I2C address 
#define SDA 23        //sda pin# on ESP32 board is 23 (some boards use 21)
#define SCL 22        //scl pin# on ESP32 board is 22
#define rotation U8G2_R0 //flawed, but I need this rotation
U8G2_SH1107_128X128_F_HW_I2C u8g2(rotation, U8X8_PIN_NONE, SCL, SDA);
void setup(void) {
  int i2cAdr = (I2C_Adr * 2);  //per u8g2 reference manual address must be multiplied by 2
  u8g2.setI2CAddress(i2cAdr);
  u8g2.begin();
  u8g2.setFont(u8g2_font_helvR10_tf);
}
void loop(void) {
  u8g2.clearDisplay();
  u8g2.drawStr(0,15,"Hello World!");
  u8g2.sendBuffer();
  delay(500);
  u8g2.drawStr(0,30,"1234567890");
  u8g2.sendBuffer();
  delay(1000);
}
----------------------------------------
B.t.w. "1234567890" displays like "567890      1234"

I uploaded a shortened version of Adafruits "Hello World" test sketch onto the ESP32 board and found that the text is displayed correctly. So it seems to me that the problem is in U8G2 for the SH1107 chip.

I would like to stick with u8g2 for the power save mode (screen saver).

chauens, 2024-04-23:
I need to add: the option to define SDA and SCL pins is not an issue. Just connect the OLED where Adafruit wants it. But I want to be able to use the "screen saver" which Adafruit does not seem to have.

olikraus, 2024-04-24:
The reference position for the string is the lower left corner of the glyphs (baseline) as you can see in the example image here: https://github.com/olikraus/u8g2/wiki/u8g2reference#drawstr

This can be change with the font-position functions: https://github.com/olikraus/u8g2/wiki/u8g2reference#setfontpostop

Wrap around effects can be avoided by putting u8g2 into 16 bit mode (see FAQ https://github.com/olikraus/u8g2/blob/master/doc/faq.txt). 16 Bit mode is automatically activated with ESP32.

chauens, 2024-04-26:
Using the Adafruit_SH110X.h and Adafruit_GFX.h libraries provides the solution to the problem as well as a "screen saver" as I just found out. So I wil go this route. 
I leave this open for now for you to decide wether this can be fixed. I don't think I'm doing anything wrong or missed something. See my previous comment. Thank you

xray-bit, 2024-05-08:
> Using the Adafruit_SH110X.h and Adafruit_GFX.h libraries provides the solution to the problem as well as a "screen saver" as I just found out. So I wil go this route. I leave this open for now for [...]

Hi, @chauens. I got the same problem.

I found a comment here:  
https://github.com/olikraus/u8g2/blob/3e86287a55e819bc8be66a5c2674f07e5e140ff1/csrc/u8x8_d_sh1107.c#L466-L469

> 128x128 OLED: this display has a very strange x offset

xray-bit, 2024-05-08:
Hi, @chauens, I got the same problem.

the default display config for sh1107_128x128 has a big x offset (check L519-520): 
https://github.com/olikraus/u8g2/blob/3e86287a55e819bc8be66a5c2674f07e5e140ff1/csrc/u8x8_d_sh1107.c#L501-L523

you can try to use pimoroni's config or seeed's config :
https://github.com/olikraus/u8g2/blob/3e86287a55e819bc8be66a5c2674f07e5e140ff1/csrc/u8x8_d_sh1107.c#L600-L622

I don't use Arduino, but I think you can try to use `U8G2_SH1107_PIMORONI_128X128_F_HW_I2C` instead of `U8G2_SH1107_128X128_F_HW_I2C` for initialization

chauens, 2024-05-09:
Hi, thank you for the tip, that appears to be very helpful. I will try that. 
Meanwhile I went back to using a 128x64 display because it actually is physically wider. I would need to use a smaller font to display the same number of characters per line on the 128x128 OLED... at least on the one I got, making it harder to read. 
So it may take some time until I get around to do another test with 128x128 OLED.

dcmrobin, 2025-01-09:
> Hi, @chauens, I got the same problem.
>
> the default display config for sh1107_128x128 has a big x offset (check L519-520): [...]

This fixed my issue. I was using U8G2_SH1107_128X128_F_4W_HW_SPI before, but then I added PIMORONI in the middle and U8G2_SH1107_PIMORONI_128X128_F_4W_HW_SPI made it work now. Hooray!

chauens, 2025-02-19:
Hi dcmrobin,
thank you for your reply. Yes I was able to work around the problem by using the "pimoroni". 
However, it would appear to be desirable that the owner of the code would provide a fix for this.