olikraus/u8g2 #2436: Support for Monochrome Low-Power ST7302 Display

korsehindi, 2024-04-27:
[image]

Hi
The display doesn't need a backlight so it's very low power, typically about 30µA at 3.3V, which is ideal for battery-powered projects. For comparison a colour TFT display typically takes 20mA.
Please add the ST7302 display to your library.

[Datasheet](https://github.com/zhcong/ST7302-for-arduino/blob/main/doc/ST7302_V0.0.pdf)
More info [+](http://www.technoblogy.com/show?3YB0),[+](https://github.com/zhcong/ST7302-for-arduino)

Thanks.

olikraus, 2024-04-27:
Looks complicated.
I will try to order one of those displays: https://de.aliexpress.com/item/1005003982915930.html
but I have to pospone this oder by a week or so...

olikraus, 2024-08-30:
Finally I have ordered this display.

korsehindi, 2024-08-31:
> Finally I have ordered this display.

Thank you for your effort, I look forward to it.

martino87r, 2024-09-16:
Apparently this is a very good display according to several reviews, so I gave up and ordered 5 pcs to play with :D 
I'm still waiting for the delivery, meanwhile would be interested on the status of the support for it since I've seen there is a lot of commits in the last few days.

olikraus, 2024-09-17:
Regarding the power consumption: This seems to be a "reflective LCD". Such displays do not contain a backlight LED. Without backlight available or activated, any LCD will consume current far below 1mA. 
However indeed this display has some extra low power features.

Regarding the progress: Yes I started to setup the u8g2 device issues/solutions so far:
- The startup sequence from github did contain some commands, which are not mentioned and are not inline with the datasheet 
- The startup sequence from github had caused some startup issues in u8g2, so I had to play with the timing. this is also not yet fully solved
- The memory architecture of that display is the most critical difference to most other displays supported by u8g2: There will be no hardware flip (U8G2_x rotation is still possible) and also no support for u8x8 
- The memory architecture also has forced to use portrait mode by default
- The memory architecture at the moment has forced a lot of bit level operation, which slows down the performance, additionally there is a two pixel gap on the left side

So open problems are:
- Fix the two pixel gap on the left side
- Analyse to which extend the bit level operations can be optimized
- Find a better and reliable startup sequence to benefit from the low power features
- Provide a list of commands for the users to balance speed versus power consumption

This is the current (draft) pixel conversion procedure, which probably requires optimization:
https://github.com/olikraus/u8g2/blob/8cfcc70c35c9c82f6e4723b926401e17053a4152/csrc/u8x8_d_st7301.c#L86-L212

olikraus, 2024-09-18:
Ultra low power mode: 
``` C++
  u8g2.sendF("caac", 0xb2, 0, 0, 0x39);  // low power mode 
```

olikraus, 2024-09-18:
[image]

olikraus, 2024-09-18:
The new constructor (https://github.com/olikraus/u8g2/wiki/gallery#18-sep-2024-st7302-122x250-lcd) will be available with 2.35.30

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

closed 2024-09-18 by olikraus

NoRolex, 2025-05-31:
Hello!
I just acquired a similar display, the only difference is that this one is a higher (186x384) resolution display. It works to the point where text shows up but the resolution mismatch makes it unreadable and  the not defined pixels are having an ant party. Can somebody make some changes so this can work with the existing ST7302 controller? Im not really skilled enough to make these changes to this great library.

NoRolex, 2025-06-10:
I could not find a datasheet for this specific display but I don't think it's much different from the 122x250 display.
The aliexpress [link](https://www.aliexpress.com/item/1005007694690515.html?pdp_ext_f=%7B%22sku_id%22%3A%2212000041979732801%22%7D&sourceType=1&spm=a2g0o.wish-manage-home.0.0#nav-description) i bought it from.

Constructor i used with an ESP32 Wroom: `U8G2_ST7302_122X250_1_4W_HW_SPI u8g2(U8G2_R1, /* cs=*/ 16, /* dc=*/ 2, /* reset=*/ 4);  //SDA = 23 ; SCL= 18`
I just used a 24 pin FPC/FFC Adapter Board with it and here is the result of the Flimode example with the 122x250 constructor.
Here is a side by side picture from a 122x250 and a 168x384 display. (The displays are wired in parallel)

I really hope it will be an easy solution :)

[image]
[image]
[image]
[image]

olikraus, 2025-06-10:
Thanks for the images, this helps, The aliexpress links mentions a ST7305, datasheet probably is this https://admin.osptek.com/uploads/ST_7305_V0_2_d0b99d9cdb.pdf

I will look into this a little bit deeper.

NoRolex, 2025-06-12:
I think i seen this display on alibaba a while ago the only problem is that a lot of the shops needs a minimum order of a 1000 pcs. 
Only found [this](https://www.alibaba.com/product-detail/2-9-Inch-White-Black-Color_1601039757990.html?spm=a2700.galleryofferlist.normal_offer.d_title.1f0d13a0GrYZIe) this seller that might be promising if you are interested in it.

The code you sent shows this result with the prevoiusly mentioned constructor:
I added a video to better demonstrate the flickering. Maybe a resolution mismatch causes this?
I also noticed that on the "noisy" part of the display it can get certain amount of grayscale. Is this a feature of this controller or is it just simply a thing that somehow happens?

https://github.com/user-attachments/assets/7c22b691-972c-4141-a3fc-711ec26baa73
[image]
[image]

olikraus, 2025-06-12:
Thanks for the images. Like you also mentioned, there are many issues to solve.

Thanks for the alibaba link, however i have never ordered something from alibaba. I probably also need a FPC connector, which then also requires some external elements and connections. Sounds like too much of effort for me at the moment.

ischenz, 2025-06-13:
Thank you for your discussion. I also particularly need U8G2 to support ST7305.