olikraus/u8g2 #2441: U8g2 font size and type

mario44, 2024-05-07:
I created Polish characters like "ĄĆĘŁŃÓŹŻąćęłńóśźż" and everything works very well after creating u8g2. My question is: Is it possible to create larger or smaller fonts using your site? If not, do you plan to introduce the size and type of fonts you create in the future? Thank you for the wonderful U8G2 font creator.

olikraus, 2024-05-10:
U8g2 just includes existing fonts from the internet. 

It is a fundamental property of the existing bitmap fonts, that they are handcrafted for a specific size.

Moreover, not always all Polish chars are included. You probably need to search in the existing u8g2 font collection. This page could be a starting point: https://github.com/olikraus/u8g2/wiki/fntgrp

mario44, 2024-05-11:
I am planning to write a program for Windows that will be able to read all characters of a contained font from a file, e.g. font.c. Is there a chance to share more detailed file data, such as the header and the structure of the saved glyphs. This description on your website is not very clear to me, I need a byte-by-byte description and how to read it, is it necessary to convert ASCII codes to HEX or maybe OCT??? Thanks again for the very useful u8g2lib library. Regards, Mario.

olikraus, 2024-05-11:
https://github.com/olikraus/u8g2/wiki/u8g2fontformat