olikraus/u8g2 #2450: U8g2 library issue with Arduino nano RP2040 processor

Herwig9820, 2024-05-28:
Hi, could it be that the U8g2 does not support the Arduino nano RP2040 ?

With a nano 33 IoT, the code below works perfectly.
But with the RP2040, I notice that MOSI and CLOCK do not emit anything (I measure a constant 3.3 Volt, with a scope) but the RESET pin constantly emits a waveform, as soon as the U8g2 library is initialized. 
The pinout of both processors is the same, as far as the used pins are concerned. So in my humble opinion the issue is somewhere in the library. Is that a possibility ?

My code snippet:
```
constexpr int VMA437_OLED_DC_PIN{ 17 };         // OLED data pin
constexpr int VMA437_OLED_CLK_PIN{ 20 };        // OLED clock pin
constexpr int VMA437_OLED_MOSI_PIN{ 21 };       // OLED MOSI pin

U8X8_SH1106_128X64_NONAME_4W_SW_SPI u8x8_spi(VMA437_OLED_CLK_PIN, VMA437_OLED_MOSI_PIN, VMA437_OLED_CS_PIN, VMA437_OLED_DC_PIN);  // SW SPI

U8X8LOG u8x8log_spi;                                                                // create U8x8LOG object                                                                        
uint8_t u8log_buffer_spi[U8LOG_WIDTH * U8LOG_HEIGHT];                   // allocate display memory 

void setup() {
    Serial.begin(115200);
    delay(5000);

    u8x8_spi.begin();                                                       
    u8x8_spi.setFont(u8x8_font_8x13_1x2_f);                                 
    u8x8log_spi.begin(u8x8_spi, U8LOG_WIDTH, U8LOG_HEIGHT, u8log_buffer_spi)
    u8x8log_spi.setRedrawMode(0);	

    u8x8log_spi.println("OLED(SW SPI) OK");
...	
}                         

```

Thanks for looking into this.
Herwig Taveirne

Herwig9820, 2024-05-29:
I checked them now. Issue 2073 seems to describe a similar problem.

olikraus, 2024-06-16:
Not sure how to support here, but did you check some of the already closed issues, like this https://github.com/olikraus/u8g2/issues/1928 or this https://github.com/olikraus/u8g2/issues/2385 ?

Herwig9820, 2024-06-21:
Hi, I'm not going to invest more time in it. Switched back to a nano ESP32 - and your library works perfectly. I'm closing the issue now.

closed 2024-06-21 by Herwig9820