olikraus/u8g2 #2453: Can I transmission content length max than 32 byte?

WangShuoran, 2024-06-04:
I know the u8g2 can not translate bigger than 32 byte data, but I want to use I2C DMA transmission OLED info.

I want to let u8g2 translate one frame info bigger than 32 byte, I think 128 byte as a frame.

How can I change this?

olikraus, 2024-06-16:
As far as I remember the 32 byte limit comes from the Arduino Framework.
The limit itself is hard coded and in fact it is reduced further to be on the safe side in u8g2, see fo example here:
https://github.com/olikraus/u8g2/blob/c7497ac0ba9697b64f2927bb6d6593682a494aa1/csrc/u8x8_cad.c#L579

Of course you can increase that value, but you need to identify the correct CAD procedure in u8x8_cad.c and you also need to ensure not to call any Arduino specific I2C transfer method.

olikraus, 2024-06-22:
The limitations had been introduced because of the Arduino IDE.
Indeed the limits are handled at different places in the software. All in all it will be difficult to change this.