olikraus/u8g2 #2455: [u8x8.setContrast] does not work / Please help me.

SeongJongKwak, 2024-06-12:
Reference link 1: [How to set contrast on a 128x128 px (ssd1327) oled display 1](https://forum.arduino.cc/t/how-to-set-contrast-on-a-128x128-px-ssd1327-oled-display/582049)
Reference link 2: [SSD1306 u8x8.setContrast(0-255) aint working · Issue #686 · olikraus/u8g2 · GitHub 1](https://github.com/olikraus/u8g2/issues/686)

MCU : STM32Duino STM32F103C8

U8X8_SSD1306_128X32_UNIVISION_SW_I2C u8x8
[u8x8.setContrast] does not work.

hello.
Other operations work fine using the 'u8x8' library.
I want to use it with low power.

Software case(1)
`u8x8.setContrast(0);`

Software case(2)
```
  u8x8_cad_StartTransfer(u8x8.GetU8x8());
  u8x8_cad_SendCmd( u8x8.GetU8x8(), 0xD9);
  u8x8_cad_SendArg( u8x8.GetU8x8(), 0);  
  u8x8_cad_EndTransfer(u8x8.GetU8x8());
```

I use the command above, but 0 and 255 make no difference.

Is there a software solution?
does not work.
In terms of hardware, I replaced 'R5' in the picture with a resistor as large as possible.
ex) R5 = 3M[Ohm]
Neither 1 nor 2 affect brightness.
help me.

olikraus, 2024-06-16:
There are many more issues related to this topic, just mentioning some of them:

https://github.com/olikraus/u8g2/issues/636

https://github.com/olikraus/u8g2/issues/1833

All in all this is very hardware specific and has nothing todo with u8g2 as a library.

SeongJongKwak, 2024-06-16:
@olikraus

Thanks for your reply.

closed 2024-08-21 by olikraus