olikraus/u8g2 #2460: New display st7571 128x128 with 3 wire SPI

bnv12345, 2024-06-17:
Hello. I bought this display from the Chinese.

[image]
[image]
[image]

Added a couple of lines of code to the file  - u8x8_d_st7571.c

    case U8X8_MSG_DISPLAY_DRAW_TILE:
      u8x8_cad_StartTransfer(u8x8);

      x = ((u8x8_tile_t *)arg_ptr)->x_pos;    
      x *= 8;
      x += u8x8->x_offset;
      u8x8_cad_SendCmd(u8x8, 0x010 | (x>>4) );                                          // Set Column Address (MSB) 
      u8x8_cad_SendCmd(u8x8, 0x000 | ((x&15)));                                         // Set Column Address (LSB) 
      u8x8_cad_SendCmd(u8x8, 0x0b0 | (((u8x8_tile_t *)arg_ptr)->y_pos));                // Set Page Address
    
      u8x8_cad_SendCmd(u8x8, 0x0E8);                                                    // Set Display Data Length (DDL) command - 11101000 
      u8x8_cad_SendCmd(u8x8, 0x080);                                                    // Display Data Length (bytes) - 128
      
      do
      {
        c = ((u8x8_tile_t *)arg_ptr)->cnt;
        ptr = ((u8x8_tile_t *)arg_ptr)->tile_ptr;
        /* SendData can not handle more than 255 bytes */
	/*
        if ( c > 31 )
        {
          u8x8_cad_SendData(u8x8, 31*8, ptr); 
          ptr+=31*8;
          c -= 31;
        }
	*/
        
        u8x8_cad_SendData(u8x8, c*8, ptr); 	
        arg_int--;
      } while( arg_int > 0 );

      
      u8x8_cad_EndTransfer(u8x8);
      break;

I added these two lines:
 u8x8_cad_SendCmd(u8x8, 0x0E8);                                                    // Set Display Data Length (DDL) command - 11101000 
 u8x8_cad_SendCmd(u8x8, 0x080);                                                    // Display Data Length (bytes) - 128

Everything worked great
[image]
Can you add a constructor under this display? I think it should look like this:
 U8G2_ST7571_128X128_F_3W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* reset=*/ 8);

olikraus, 2024-06-19:
Which constructor did you use as a starting point?

bnv12345, 2024-06-19:
U8G2_ST7571_128X128_F_4W_HW_SPI u8g2

bnv12345, 2024-06-19:
Yes, I didn’t even think about the u8x8 mode.

olikraus, 2024-06-20:
Note to myself: We need a new 3W CAD procedure and a new interface for the code generator, the above command should be sent in the CAD procedure.

olikraus, 2024-08-30:
not sure how to continue from here, maybe I need to reconfigure an existing ST7571 LCD in my lab and test the 3-wire SPI.