olikraus/u8g2 #2461: U8X8_SSD1327_WS_128X128_HW_I2C blocks loop

ruediheimlicher, 2024-06-17:
I have a program on Atmega328p with platformio and am trying  to connect an OLED display with SSD1327 driver.

I include the libs from olikraus:

```
#include <U8g2lib.h>
#include <U8x8lib.h>
```
and set 

`U8X8_SSD1327_WS_128X128_HW_I2C u8x8(A4,A5);`
That compiles well, but 

`u8x8.begin();`

blocks the loop.

My code:

``` 
#include <Arduino.h> 
 #include <avr/io.h>
 #include <util/delay.h>
#include <avr/interrupt.h>
#include <avr/pgmspace.h>
#include <inttypes.h>
#include "defines.h"
#include <stdint.h>

#include "lcd.c" // LCD-Stuff

#include <U8g2lib.h>
#include <U8x8lib.h>

U8X8_SSD1327_WS_128X128_HW_I2C u8x8(A4,A5); // ok

#define LOOPLEDPORT		PORTD
#define LOOPLEDDDR      DDRD
#define LOOPLED			5

volatile uint8_t   loopstatus=0x00;            

uint16_t loopcount0=0;
uint16_t loopcount1=0;
uint16_t loopcount2=0;

uint16_t refreshtakt = 0x1FF;

void slaveinit(void)
{   
   LOOPLEDDDR |= (1<<LOOPLED);

	//LCD
	LCD_DDR |= (1<<LCD_RSDS_PIN);	//Pin 5 von PORT B als Ausgang fuer LCD
 	LCD_DDR |= (1<<LCD_ENABLE_PIN);	//Pin 6 von PORT B als Ausgang fuer LCD
	LCD_DDR |= (1<<LCD_CLOCK_PIN);	//Pin 7 von PORT B als Ausgang fuer LCD

}  
    
int main (void) 
{
	slaveinit();  
 	// initialize the LCD 
	lcd_initialize(LCD_FUNCTION_8x2, LCD_CMD_ENTRY_INC, LCD_CMD_ON);
	lcd_puts("A328_PIO_Start");
	
	_delay_ms(200);
       u8x8.begin(); // blocks loop
       _delay_ms(200);
   
       sei();
	while (1)
       {  
         loopcount0++;
         if (loopcount0>=refreshtakt)
         {
            loopcount0=0;
            loopcount1++;
            if (loopcount1>=refreshtakt)
            {
               loopcount1 = 0;
               LOOPLEDPORT ^= (1<<LOOPLED); 
               lcd_gotoxy(0,1);
               lcd_putint(loopcount2);
               loopcount2++;
            }           
            loopcount0=0;
         }  // loopcount0>=refreshtakt
   }//while
 return 0;
}
```

olikraus, 2024-06-19:
Maybe the pull up resistors for the I2C bus are missing?

ruediheimlicher, 2024-06-19:
pull-ups are present.

I also tried with a adafruit 1306 128x64-display that works fine with an uno and arduino. 
Same result: loop blocked.

Gruss
Ruedi

olikraus, 2024-06-20:
I think the constructor is wrong: `U8X8_SSD1327_WS_128X128_HW_I2C u8x8(A4,A5);`
Actually A4 and A5 are fixed for most of the controllers, so it is not needed for u8g2, instead the one and only argument should be the reset line (which might not be there in your case). So try this instead:

`U8X8_SSD1327_WS_128X128_HW_I2C u8x8(U8X8_PIN_NONE);`

See also here: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#communication

ruediheimlicher, 2024-06-22:
Tried this constructor.
result:
- no more blocking
- some pixels on the screen visible, 
- I attach 2 packages
I let the A328 send a 'A' every 500ms. In the Analyser, I see the data packages, but they are not all the same. 
I attach 2 packages and a sequence what happens on the screen.
[image]
[image]

https://github.com/olikraus/u8g2/assets/1231384/c6dc86f8-5d4d-4c78-881d-4b323a9c8270

olikraus, 2024-06-23:
I don't understand your initial code. Maybe it is better to just use one of the U8g2 examples, which are available in the Arduino IDE (like u8g2 HelloWorld.ino)
Just add the constructor into the example and check the result. How will it look?

ruediheimlicher, 2024-06-25:
Success!
After setting the address to 2*0x3D (was 0x3C) and reducing bus clock to 400000 the display writes as expected.
Thanks for your advices.

closed 2024-08-21 by olikraus