olikraus/u8g2 #2462: ST7567S I2C unknown variant

kittenarmy, 2024-06-22:
I have an ST7567S variant I2C display that looks identical to [this](https://github.com/olikraus/u8g2/issues/2364), it has I2C address 0x3F. I tried using these 4 as per thread but none of them work with the Hello World example: 

- U8G2_ST7567_JLX12864_1_SW_I2C
- U8G2_ST7567_ENH_DG128064_1_SW_I2C
- U8G2_ST7567_ENH_DG128064I_1_SW_I2C
- U8G2_ST7567_OS12864_1_SW_I2C

Just to make sure, how should I connect it to Arduino Nano with this library? I have the display connected to I2C pins A4, A5 and putting `U8G2_ST7567_OS12864_1_SW_I2C u8g2(U8G2_R0, 19, 18);` where it expects the constructor.

kittenarmy, 2024-06-22:
If I use GraphicsTest.ino with `U8G2_ST7567_HEM6432_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);` I get partially working (screen seems to be far too bright though):

[image]

kittenarmy, 2024-06-22:
I don't know if it helps but [this library](https://github.com/luetee/ST7567S_128X64_I2C/) mostly works (except I can't get it to display images). There is a datasheet included.

olikraus, 2024-06-23:
> Just to make sure, how should I connect it to Arduino Nano with this library?

In case of SW_I2C you can use any kind of pin. Just provide the pin numbers for clock and data.

> but none of them work with the Hello World example

Sometimes it is better to use the Contrast example, because the displays might differ in the contrast setup (u8g2.setContrast() might be required)

kittenarmy, 2024-06-23:
> In case of SW_I2C you can use any kind of pin. Just provide the pin numbers for clock and data.
>
>
> Sometimes it is better to use the Contrast example, because the displays might differ in the [...]

Thanks. I tried each within ContrastTest.ino and some seem to alter the contrast a little. But when I try these constructor in GraphicsTest.ino I get nothing.

Is this library compatible with Arduino 1.8.19? I used the latest available from the manager.

```
// for reference
// nano sda grey a4, scl white a5
// 19 clock, 18 data
// U8G2_ST7567_JLX12864_1_SW_I2C u8g2(U8G2_R0, A5, A4);
// U8G2_ST7567_OS12864_1_SW_I2C u8g2(U8G2_R0, A5, A4);
// U8G2_ST7567_ENH_DG128064_1_SW_I2C u8g2(U8G2_R0, A5, A4);
// U8G2_ST7567_ENH_DG128064I_1_SW_I2C u8g2(U8G2_R0, A5, A4);

// some contrast changing in ContrastTest.ino
// U8G2_ST7567_PI_132X64_1_HW_I2C u8g2(U8G2_R0);
// U8G2_ST7567_PI_132X64_1_2ND_HW_I2C u8g2(U8G2_R0);
```

olikraus, 2024-07-21:
u8g2 should be compatible with 1.8.19, but I am a little bit behind testing the latest arduino version. In fact also Arduin 2.x should work.

olikraus, 2024-08-21:
I have Arduino 1.8.19 on my laptop, so far it's working fine.

kittenarmy, 2024-09-01:
Is there a way to modify your library so it works with my display variant? I can only get it to work properly with another library.

olikraus, 2024-09-01:
hmm... you mentioned, that your display looks like the one in #2364 
so  U8G2_ST7567_ENH_DG128064I_1_SW_I2C should work.
So this is not the case?

olikraus, 2024-09-01:
Anyhow, I would need much more information about your display.
You did not even mention the resolution.