olikraus/u8g2 #2466: Bdfconv won't run on Win 11 machine

No1sonuk, 2024-06-30:
As title:
I can't run bdfconv.exe on my Win 11 64-bit computer.
Error message says it's not compatible with 64-bit.

olikraus, 2024-07-21:
oh, ok, I will try to verify this on my own win system.

olikraus, 2024-07-21:
I can confirm, that it work fine on my win environment

No1sonuk, 2024-07-21:
[image]
Still doesn't work on mine...

No1sonuk, 2024-07-21:
[image]
Still doesn't work on mine.

olikraus, 2024-08-18:
I am not a windows expert, so it's difficult to tell what is the problem here. 
I have create the binary with mingw environment, so maybe as a workaround, just install msys2 and mingw and compile the code by yourself

closed 2024-08-29 by olikraus