olikraus/u8g2 #2468: API Functions for a 'back' button

microbmen, 2024-07-05:
Hello @olikraus ,  

I have implemented an up and down button,  prevField, and nextField respectively.  
I have implemented a select button,  sendSelect, 

What I Cant figure out is how to implement a back button. 

Requirement:
API call which does a sendBack() function, returning to the previous MUI_FORM on the previous cursor position.

We can use gotoForm as a back button, and track the form ID needed depending on the current form displayed, however gotoForm requires an initial position, which gives odd UI/UX performance.  
An API call for currentCursor(), currentForm(), would also suffice, as then we could retain the position and form independent and use gotoForm anyway we'd like.  

Unless I am missing something, this seems to be a needed feature for 4 button control (up/down , previous/next) 

Thanks!

qianfan-Zhao, 2024-07-16:
I am learing 'mui' example this days. All the menu entry has a ui return button such as this example:

```c
/* number entry demo */
MUI_FORM(10)
MUI_STYLE(1)
MUI_LABEL(5,10, "Number Menu")
MUI_XY("HR", 0,13)
MUI_STYLE(0)

MUI_LABEL(5,27, "Number [mse]:")
MUI_XY("IN",76, 27)
MUI_LABEL(5,41, "Number [mud]:")
MUI_XY("IH",76, 41)

MUI_XYAT("G1",64, 59, 0, " OK ")
```

The last line `MUI_XYAT` has a pointer value set which form will return to when this button is pressed, seems all the menu entries use this button to return to it's parent, I can not find an example which use hardware 'backup' button handle it.

I am interesting how to add a hardware backup button, maybe we can take more discussion.

qianfan-Zhao, 2024-07-19:
@microbmen Hi, I pushed a new PR #2479 try add back button, could you please try it?

microbmen, 2024-07-19:
ok, will try it.  i created a work around by setting a 'current menu' flag using MUI_AUX and then using gotoform.

olikraus, 2024-07-21:
Currently I do hesitate to include the above mentioned PR, because there is already a little bit of code, which should support this. I also think, that the cursor topic is not addressed by pr #2479 .
Let me look into this a little bit deeper.

olikraus, 2024-07-21:
Let me give a small background on this topic. There are two concepts in u8g2:
1. There is a cache concept, which stores the cursor position along with the form id. The idea is, that if you jump to a specific form, then the cursor position is automatically restored. However the current cache size is very small but could be easily extended: https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/csrc/mui.h#L226
Maybe your problem is already solved by increasing this value

2. Other concept is an automatic jump back to the previous form and cursor position. At the moment there is just one level of jump back, which is used by the parent / child menus (like the combo menu). So basically this feature is not available for your app itself. 

To me the question is: what is the real intention of your initial question. I feel it is more like a back to Home menu jump, regardless from where. This should work with a normal jump by using https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/csrc/mui.h#L226
and increasing the above mentioned cache.

microbmen, 2024-07-22:
The intension is this,  suppose I traverse from MUI_FORM(1) to MUI_FORM(3), to MUI_FORM(7),  
and MUI_FORM(1) is a list of 4 submenus, I select the 2nd item which takes me to MUI_FORM(3), MUI_FORM(3) has 4 item list.  I select the 4th item, which takes me to MUI_FORM(7).
I want the back button on MUI_FORM(7) to take me to MUI_FORM(3), and position is the 4th item.  issuing 'back' again and I would be on MUI_FORM(1) on the 2nd item.  

I've accomplished this by using variable for every single form menu, and when the back button is tapped the FORM and Initial position is set depending on that variable.  Its working, but would be nicer if 'back' was available.

qianfan-Zhao, 2024-07-22:
@olikraus @microbmen Hi, new version based on MUI_MENU_CACHE_CNT feature is pushed on #2479, please review again.

rproesel, 2024-07-24:
I was also looking for a back button, so I tried the code you posted.  It seems to work and does just what I needed. Thanks!

olikraus, 2024-08-18:
I will look into the PR once I have a little bit more time :-)

olikraus, 2024-08-25:
I started to work on a complete new solution: A back button, which allows to jump back to a previous form id.
I hope this will solve all problems.

olikraus, 2024-08-25:
https://github.com/olikraus/u8g2/wiki/muiref#back-button

olikraus, 2024-08-28:
I have added the code into the latest beta... just in case some wants to look into this...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2024-09-01:
no further feedback, closing...

closed 2024-09-01 by olikraus