olikraus/u8g2 #2477: (Proposal) Optimization for draw XBMP

v-ivanyshyn, 2024-07-13:
Current implementation of `void u8g2_DrawHXBMP(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t len, const uint8_t *b)` renders XBM image pixel-by-pixel using `u8g2_DrawHVLine()` function. It may be optimized to render horizontal line not of 1px width, but until current bit from the image remains the same. Here is the code patch:

replace:
```
    if( current_bit ) {
      u8g2->draw_color = color;
      u8g2_DrawHVLine(u8g2, x, y, 1, 0);
    } else if( u8g2->bitmap_transparency == 0 ) {
      u8g2->draw_color = ncolor;
      u8g2_DrawHVLine(u8g2, x, y, 1, 0);
    }
   
    x++;
    mask <<= 1;
    if ( mask == 0 )
    {
      mask = 1;
      b++;
    }
    len--;
```
with:
```
    int run_length = 0;
    // Determine the run length of consecutive bits with the same color
    while (len > 0 && ((u8x8_pgm_read(b) & mask) == current_bit))
    {
        run_length++;
        x++;
        mask <<= 1;
        if (mask == 0)
        {
            mask = 1;
            b++;
        }
        len--;
    }
    if (current_bit)
    {
        u8g2->draw_color = color;
        u8g2_DrawHVLine(u8g2, x - run_length, y, run_length, 0);
    }
    else if (u8g2->bitmap_transparency == 0)
    {
        u8g2->draw_color = ncolor;
        u8g2_DrawHVLine(u8g2, x - run_length, y, run_length, 0);
    }

```
I've tested it on STM32F103 with 2 images:
[image]
101x61px
performance using current approach: 23730 us
performance using proposed approach: 12537 us

[image]
94x60px
performance using current approach: 21430 us
performance using proposed approach: 16454 us

Looks like pretty solid improvement in performance. All credits for the solution to ChatGPT

olikraus, 2024-08-18:
Nice, maybe you can paste the complete function

v-ivanyshyn, 2024-08-22:
Here's how it looks in my code:

```
void u8g2_DrawHXBMP(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t len, const uint8_t *b)
{
  uint8_t mask;
  uint8_t color = u8g2->draw_color;
  uint8_t ncolor = (color == 0 ? 1 : 0);
#ifdef U8G2_WITH_INTERSECTION
  if ( u8g2_IsIntersection(u8g2, x, y, x+len, y+1) == 0 ) 
    return;
#endif /* U8G2_WITH_INTERSECTION */
  
  mask = 1;
  while(len > 0)
  {
    uint8_t current_bit = u8x8_pgm_read(b) & mask;
#if 0
    if( current_bit ) {
      u8g2->draw_color = color;
      u8g2_DrawHVLine(u8g2, x, y, 1, 0);
    } else if( u8g2->bitmap_transparency == 0 ) {
      u8g2->draw_color = ncolor;
      u8g2_DrawHVLine(u8g2, x, y, 1, 0);
    }
   
    x++;
    mask <<= 1;
    if ( mask == 0 )
    {
      mask = 1;
      b++;
    }
    len--;
#else
    int run_length = 0;
    // Determine the run length of consecutive bits with the same color
    while (len > 0 && ((u8x8_pgm_read(b) & mask) == current_bit))
    {
        run_length++;
        x++;
        mask <<= 1;
        if (mask == 0)
        {
            mask = 1;
            b++;
        }
        len--;
    }
    if (current_bit)
    {
        u8g2->draw_color = color;
        u8g2_DrawHVLine(u8g2, x - run_length, y, run_length, 0);
    }
    else if (u8g2->bitmap_transparency == 0)
    {
        u8g2->draw_color = ncolor;
        u8g2_DrawHVLine(u8g2, x - run_length, y, run_length, 0);
    }
#endif
  }
  u8g2->draw_color = color;
}
```
Probably it would make sense to introduce #if flag for this optimization as it's done with other features, but I haven't done it on my side (that's why posted it as an issue rather to a pull request).

olikraus, 2024-09-08:
hmmm there is an issue. Your code will do only the run length encoding if the current bit is zero.
I think the correct version should be:
``` C++
while (len > 0 && (current_bit == 0 ? ((u8x8_pgm_read(b) & mask) == 0) : ((u8x8_pgm_read(b) & mask) != 0 )  ))
```

olikraus, 2024-09-08:
Updated code:
https://github.com/olikraus/u8g2/blob/a0005cb6b8da590d94ea98eb93fe5f6bf0520d16/csrc/u8g2_bitmap.c#L192-L252

olikraus, 2024-09-08:
Again, thanks for the input!

closed 2024-09-08 by olikraus