olikraus/u8g2 #2480: drawCircle() artifacts if used with inverse drawing

DatBlaueHus, 2024-07-23:
If using draw Circle with inverse Drawing, I get some artifacts, both the lines and the discs expose dirk lines crossing them, looks like an off-by-one error to me.

Code to reproduce: 

`#include <U8g2lib.h>
#include <Wire.h>

//Choose the driver according to memory requirements
// U8G2_SSD1306_128X64_NONAME_1_HW_I2C display(U8G2_R2, /* reset=*/ U8X8_PIN_NONE); //small buffer
U8G2_SSD1306_128X64_NONAME_2_HW_I2C display(U8G2_R2, /* reset=*/ U8X8_PIN_NONE); // big buffer
//U8G2_SSD1306_128X64_NONAME_F_HW_I2C display(U8G2_R2, /* reset=*/ U8X8_PIN_NONE); // Fullscreen

void setup() {
  display.begin();
  drawOnce();  
}

void drawOnce() {
  display.firstPage();
  do {     //page looping
    display.setDrawColor(1);
    display.drawBox(8, 8, 102, 102);
    display.setDrawColor(2);
    display.drawCircle(91, 29, 8);
    display.drawDisc(91, 29, 4);
    display.drawCircle(120, 32, 8);
    display.drawDisc(120, 32, 4);
  } while ( display.nextPage() );
}

void loop() {
}
`

olikraus, 2024-08-18:
marked as a bug for now... I need to find some time to look into this... Thanks for reporting this.

olikraus, 2024-08-29:
The problem is the DrawColor(2). This is not supported. See here: https://github.com/olikraus/u8g2/wiki/u8g2reference#drawdisc

olikraus, 2024-09-01:
no further feedback, closing...

closed 2024-09-01 by olikraus