olikraus/u8g2 #2483: SSD1312 Controller with 120x28 pixel OLED panel

edwarepl, 2024-08-03:
I have used the u8g2 library on my STM32 project and I love it!
My OLED is based upon the SSD1312 IC,  but has only 120x28 pixels.
I can get by using the u8g2_Setup_ssd1312_i2c_128x32_f display configuration for a 132x32 OLED, but the rotation settings don't work well, and the 0,0 origin is outside of my oled window.
The display data is here:
https://goldenmorninglcd.com/oled-display/0.63-inch-oled-120x28-ssd1312-gme12028-01/

Is it possible to add a custom display size to work with u8g2, without modifying the u8g2 source? (to avoid issues when updating u8g2).
Thanks!

FengyiTian1, 2024-08-08:
We recommend modifying the source code, as this is the most straightforward and effective way to achieve your goal without losing performance. If you prefer not to modify it, you can try adjusting the origin coordinates to offset them to the appropriate position. We hope this helps!

Fengyi Tian
Global Sales Executive
fengyi.tian@goldenmorninglcd.com
Shenzhen GoldenMorning Electronic Co.,Ltd

olikraus, 2024-08-18:
Is there a way to download the display datasheet for offline reading?

FengyiTian1, 2024-08-19:
> Is there a way to download the display datasheet for offline reading?

We have now opened the download permissions for all uploaded datasheets. You can download them from our website. We are also gradually adding more datasheets. Additionally, if you have specific requirements for datasheets corresponding to certain sizes or models, you can reply here, and we will upload them to the website or send them to you individually.

olikraus, 2024-09-01:
The new constructor U8G2_SSD1312_120X32_xxx should be available now in the latest beta release.
Can someone conform, that this works?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

edwarepl, 2024-09-07:
I have reviewed the changes. It seems the changes were made allow for a 120 x 32 resolution display.
The resolution required is 120 x 28, so the updated driver doesn't suit the required resolution.

olikraus, 2024-09-07:
ah, yes, looks like I missed this. I will change it to 120x28

olikraus, 2024-09-07:
new constructor: U8G2_SSD1312_120X28

olikraus, 2024-09-07:
i have created beta 2.35.27 with the above constructor
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

olikraus, 2024-09-11:
hope this works... closing

closed 2024-09-11 by olikraus