olikraus/u8g2 #2485: MUI Scrollable list with elements

MihirGrand, 2024-08-05:
How do I make a list of elements scrollable in MUI? Currently any more items than can fit on the screen just aren't accessible.

```
const muif_t muif_list[] = {
  MUIF_U8G2_FONT_STYLE(0, u8g2_font_helvR08_tr),
  MUIF_U8G2_FONT_STYLE(1, u8g2_font_spleen6x12_mr),
  MUIF_U8G2_FONT_STYLE(2, u8g2_font_unifont_t_86),

  MUIF_RO("HR", mui_hrule),
  MUIF_RO("CY", mui_yes),
  MUIF_RO("CN", mui_no),
  MUIF_U8G2_LABEL(),
  MUIF_RO("GP", mui_u8g2_goto_data),
  MUIF_BUTTON("GC", mui_u8g2_goto_form_w1_pi),
  MUIF_BUTTON("EX", mui_u8g2_btn_exit_wm_fi),

  MUIF_U8G2_U8_MIN_MAX("NV", &work, 0, 60, mui_u8g2_u8_min_max_wm_mud_pi),
  MUIF_U8G2_U8_MIN_MAX("NW", &shortv, 0, 60, mui_u8g2_u8_min_max_wm_mud_pi),
  MUIF_U8G2_U8_MIN_MAX("NX", &longv, 0, 60, mui_u8g2_u8_min_max_wm_mud_pi),
  MUIF_U8G2_U8_MIN_MAX("NY", &laps, 0, 9, mui_u8g2_u8_min_max_wm_mud_pi),
  MUIF_EXECUTE_ON_SELECT_BUTTON("GO", mui_u8g2_btn_goto_wm_fi)
};

const fds_t fds_data[] =

  MUI_FORM(1)
    MUI_STYLE(1)
      MUI_LABEL(5, 8, "Time")
        MUI_STYLE(0)
          MUI_XY("HR", 0, 11)
            MUI_DATA("GP",
                     MUI_2 "Configure")
              MUI_XYA("GC", 5, 24, 0)
                MUI_STYLE(2)
                  MUI_XYT("EX", 120, 15, "⭠")

                    MUI_FORM(2)
                      MUI_STYLE(1)
                        MUI_LABEL(5, 8, "Time")
                          MUI_XY("HR", 0, 11)
                            MUI_STYLE(0)
                              MUI_LABEL(5, 23, "Work:")
                                MUI_LABEL(5, 36, "Short break:")
                                  MUI_LABEL(5, 49, "Long break:")
                                    MUI_LABEL(5, 64, "Laps:")
                                      MUI_XY("NV", 80, 23)
                                        MUI_XY("NW", 80, 36)
                                          MUI_XY("NX", 80, 49)
                                            MUI_XY("NY", 80, 64)
                                              MUI_STYLE(2)
                                                MUI_XYAT("GO", 120, 15, 1, "⭠");
```

So in Form 2, I have 4 labels and corresponding minmax elements. I need to add more elements to the same screen, how do I do so? It would be great if I could also add some gotoform buttons to the same list too (declared as id GC in above code)

Btw, kudos on making this library. It's extensive, feature rich, and very stable. I had a blast using it!

olikraus, 2024-08-18:
There is no scrolling as such. I mean you can not scroll multiple elements.
Instead my suggestion would be, to have a scrollable selection list (like this: https://github.com/olikraus/u8g2/wiki/muiref#list-selection-child-form-uint8-static-list) and jump to a form where you can update details for that entry.

MihirGrand, 2024-08-18:
@olikraus  Got it! I have some more questions -
1. Is there an easy way to get which form then user is currently on? Im currently using a makeshift method, keeping track of what form the user is on based on button presses, but it's not accurate, I can only guess what level of depth the user is on.
2. Is there a way to call a function when a control updates? Like when a min max value is updated? Im trying to change some settings through an mui menu, and the settings are updated to the eeprom using a function. As above, I do have a makeshift method I'm using, but it's not the best 😅.

olikraus, 2024-08-18:
1. Current Form Id

The form number is returned by "getCurrentFormId()" (ok, not sure, whether this is there in the latest release).
Looks like documentation is missing for this function... so I just added this: https://github.com/olikraus/u8g2/wiki/muiref#getcurrentformid
Let me know if the function is not present, then you need to use the latest beta.

2. Update Dependent Values

In the MUIBlink example i just always did the update of the other settings. Anyhow, the usual way would be to "derive"/"extend" the field function to react on some special messages. The problem is, that this will depend on the mud or mse mode, because the message for the value change differs depending on the mode. Example:

Let assume you have `uint8_t mui_u8g2_u8_min_max_wm_mse_pf(mui_t *ui, uint8_t msg)`

So you create a new extended function:
``` C++
uint8_t my_mui_u8g2_u8_min_max_wm_mse_pf(mui_t *ui, uint8_t msg) 
{
  uint8_t result = mui_u8g2_u8_min_max_wm_mse_pf(ui,  msg);  // call the original function
  // check, whether the message has changed the value (mse mode)
  if ( msg == MUIF_MSG_CURSOR_SELECT || msg == MUIF_MSG_VALUE_INCREMENT || msg == MUIF_MSG_VALUE_DECREMENT )
  {
    // value changed, do some action
  }
  return result;
}
```
In the muif_list you will then use your new function as callback for `MUIF_U8G2_U8_MIN_MAX`.

For mud mode, the messages are different:

``` C++
uint8_t my_mui_u8g2_u8_min_max_wm_mud_pf(mui_t *ui, uint8_t msg)
{
  uint8_t result = mui_u8g2_u8_min_max_wm_mud_pf(ui,  msg);
  // check, whether the message has changed the value (mud mode)
  if ( msg == MUIF_MSG_EVENT_NEXT || msg == MUIF_MSG_EVENT_NEXT )
  {
    if ( result != 0 ) // in mud mode, the result will indicate whether there was a changed or not
    {
    // value changed, do some action
    }
  }
}
```

Regarding your initial question, I once did something similar to modify values within an array.
More specific: How can I create a user interface to update the following arrays: 
https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/sys/sdl/mui/main.c#L101
https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/sys/sdl/mui/main.c#L103

The solution is to have one variable for the array position and two variables for the array element.
As soon as the array pos is changed, the array is updated with the values from the form 
and the new values are loaded into the form:
https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/sys/sdl/mui/main.c#L107-L133

The form itself is simple and contains only three variables (and a button to exit the menu):
https://github.com/olikraus/u8g2/blob/a5f674fedd5e75e71389e6ad2531363b654bdf76/sys/sdl/mui/main.c#L491-L503

olikraus, 2024-08-25:
I have started to make a more complex example for this question...

olikraus, 2024-08-25:
Note: This update includes mui_hline, which probably needs to be documented in muiref

closed 2024-09-02 by olikraus

MihirGrand, 2024-09-16:
@olikraus  Sorry for the late response, all the tips you provided worked flawlessly! Thank you very much!