olikraus/u8g2 #2490: SSD1363 256x128

fritolays, 2024-08-17:
Controller Datasheet:
[SSD1363.pdf](https://github.com/olikraus/u8g2/files/13622824/SSD1363.pdf)

Related Issues:
https://github.com/olikraus/u8g2/issues/2298
https://github.com/olikraus/u8g2/issues/2318

Seems this controller/resolution has come up before without proper support being implemented.
As per your comment in https://github.com/olikraus/u8g2/issues/2298 I've tried the SSD1322 constructor and it does display without the noise.
This is how the Hello World example displays using the U8G2_SSD1322_240X128_F_4W_SW_SPI constructor.
[image]

Let me know if there is anything else I can provide or test...

fritolays, 2024-08-19:
Very kind of you, let me know if I can offset the cost in any way.

DatanoiseTV, 2024-08-20:
I am also able to test, if any help is required.

olikraus, 2024-08-24:
Thanks for your offers... It's on the way, but usually requires some time...

olikraus, 2024-08-28:
[image]

olikraus, 2024-08-28:
The new constructor will look like this: U8G2_SSD1363_256X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
The code is part of the latest beta (2.35.20)...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

DatanoiseTV, 2024-08-29:
Unfortunately doesn't seem to work for me with a TW56280270A01. Datasheet attached.
[image]

[Uploading 2.7寸TW56280270A01规格书.pdf…]()

fritolays, 2024-08-29:
> The new constructor will look like this: U8G2_SSD1363_256X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=_/ 10, /_ dc=_/ 9, /_ reset=*/ 8); The code is part of the latest beta (2.35.20)...

[image]

Lovely, thanks again for the support.

olikraus, 2024-08-29:
> Unfortunately doesn't seem to work for me with a TW28642270B (05). Datasheet attached. <img alt="Screenshot 2024-08-29 at 11 50 58" width="430" src="https://private-user-images.githubusercontent.com/6614616/362626372-67989448-902d-4966-8ba8-c11a35ffdf93.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ5NDg1NzgsIm5iZiI6MTcyNDk0ODI3OCwicGF0aCI6Ii82NjE0NjE2LzM2MjYyNjM3Mi02Nzk4OTQ0OC05MDJkLTQ5NjYtOGJhOC1jMTFhMzVmZmRmOTMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjlUMTYxNzU4WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGJhYjQxN2M1ZGQwZGVjNzU0OTViNTg4MmJkMWUyZTRmZGNiNGIzNjVlOWNkN2ExMzdiMGUwYThkMzg5YWRkZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.9R4U3XNLtE0a_L_63wXBqK8csF34KDk5k3bZAut5Uuw">
> 
> [2.7寸TW56280270A01规格书.pdf](https://github.com/user-attachments/files/16797120/2.7.TW56280270A01.pdf)

maybe you need to use the F constructor

olikraus, 2024-09-01:
no further feedback, closing...

closed 2024-09-01 by olikraus