olikraus/u8g2 #2496: CH1120 OLED driver support

RSE-dev, 2024-08-30:
I have encountered an issue with the newer CH1120 driver IC, which seems to be replacing the SH1108. 
The OLED module looks absolutely the same, has the same pinout (SPI connection) and same resolution (128x160)
Despite their similarities, the CH1120 does not function correctly with the existing SH1108 code in the library.
I'm using STM32CubeIDE, the display module is connected through the SPI. The module based on SH1108 with the same setup works perfectly

I attempted to modify the existing code:
 - declared the new function in u8g2.h
	`void u8g2_Setup_ch1120_128x160_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);`
 - created the new function in u8g2_d_setup.c
	```
void u8g2_Setup_ch1120_128x160_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
	{
	uint8_t tile_buf_height;
	uint8_t *buf;
	u8g2_SetupDisplay(u8g2, u8x8_d_ch1120_128x160, u8x8_cad_001, byte_cb, gpio_and_delay_cb);
	buf = u8g2_m_16_20_f(&tile_buf_height);
	u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
	}
```
 - declared the new function in u8x8.h
	`uint8_t u8x8_d_ch1120_128x160(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);`
 - created the file u8x8_d_ch1120.c (the clone of the u8x8_d_sh1108.c with adopted functions names and modified init_seq and display_info)
 - the init_seq was modified according to the manufacturers datasheet and rocemmendations:
	```
static const uint8_t u8x8_d_ch1120_128x160_noname_init_seq[] = {
		U8X8_START_TRANSFER(), 					/* enable chip, delay is part of the transfer start */
				U8X8_C(0x0ae), 					// display off
				U8X8_CAA(0x021,0x000,0x01F), 	// Set Column Start/End Address of Display RAM for Mono mode
				U8X8_CAA(0x022,0x000,0x04F), 	// Set Row Start/End Address of Display RAM for Mono mode
				U8X8_CA(0x0a2, 0x000), 			// Set Display Start Line
				U8X8_CA(0x081, 0x02f), 			// The Contrast Control Mode Set
				U8X8_CA(0x0ac, 0x001), 			// Set Grayscale/Mono display mode ->Mono mode
				U8X8_CA(0x020, 0x000),		    // Set Memory addressing mode?????
				U8X8_C(0x0a0),					// Set Segment Re-map
				U8X8_C(0x0c8),					// Set Common Output Scan Direction
				U8X8_CA(0x0a3, 0x000),  		// Set Display Rotation 90* - NO
				U8X8_C(0x0a4), 					// Disable Entire Display OFF/ON
				U8X8_C(0x0a6), 					// Set Normal/Reverse Display
				U8X8_CA(0x0a8, 0x07f), 			// Set Multiplex Ratio
				U8X8_CA(0x0d3, 0x010), 			// Set Display Offset
				U8X8_CA(0x0d5, 0x01f), 			// Set Display Divide Ratio/Oscillator Frequency Mode Set
				U8X8_CAAAAA(0x048, 0x002, 0x093, 0x002, 0x0d8, 0x001),// Dis-charge Period
				U8X8_CAAAAA(0x049, 0x006, 0x0d9, 0x00f, 0x094, 0x01f),// Pre-charge Period
				U8X8_CA(0x04b, 0x004),			// Pre-charge strength
				U8X8_CA(0x0da, 0x000), 			// Set SEG pads hardware configuration
				U8X8_CA(0x0db, 0x040), 			// VCOM Deselect Level
				U8X8_CA(0x0ad, 0x002), 			// External or internal IREF Set
		//		U8X8_C(0x0af),
				U8X8_END_TRANSFER(),			 /* disable chip */
				U8X8_END() 						/* end of sequence */
		};
```
 - the display_info was modified according to the manufacturers datasheet:
	```
static const u8x8_display_info_t u8x8_ch1120_128x160_noname_display_info = {
	/* chip_enable_level = */0,
	/* chip_disable_level = */1,
	
	/* post_chip_enable_wait_ns = */12, /* ch1120: 12 us tcsh */
	/* pre_chip_disable_wait_ns = */45, /* ch1120: 12 us tcss */
	/* reset_pulse_width_ms = */1, /* ch1120: 10 us */
	/* post_reset_wait_ms = */1,
	/* sda_setup_time_ns = */20, /* ch1120: 20ns tsds */
	/* sck_pulse_width_ns = */100, /* ch1120: 100ns */
	/* sck_clock_hz = */4000000UL, /* since Arduino 1.6.0, the SPI bus speed in Hz. Should be  1000000000/sck_pulse_width_ns */
	/* spi_mode = */0, /* active high, rising edge */
	/* i2c_bus_clock_100kHz = */4,
	/* data_setup_time_ns = */40, /* ch1120: 40 ns tds8 */
	/* write_pulse_width_ns = */150, /* ch1120: tCCLW */
	/* tile_width = */16,
	/* tile_height = */20,
	/* default_x_offset = */16,
	/* flipmode_x_offset = */16,
	/* pixel_width = */128,
	/* pixel_height = */160 };
```
	
Unfortunately, changes I made didn't lead to the working module, only displaying "the snow". But the display reacts for the setting the contrast, rotation and u8g2_SetPowerSave

Please find here the datasheet for the OLED driver IC and for the OLED display based on it
[WEA160128BWPP3N00000.pdf](https://github.com/user-attachments/files/16816311/WEA160128BWPP3N00000.pdf)
[CH1120 V0.9.pdf](https://github.com/user-attachments/files/16816312/CH1120.V0.9.pdf)

olikraus, 2024-09-03:
It looks like that the CH1120 is a graylevel display, while SH1108 is a pure monochrome display. 
At least CH1120 needs to be configured as a monochrome display, see chapter 6.7.3 of the datasheet.

Did you compare the command table of both display? I think there are not so many similarities... Anyhow I will follow your apprach and create a copy of the SH1108 code...

olikraus, 2024-09-03:
I tookover the init sequence from the display datasheet. It is very much different to the SH1108. I really wonder that anything was visible with the SH1108 constructor.

I have created beta 2.35.25 with constructor U8G2_CH1120_128X160_1_4W_HW_SPI and friends

Maybe you can test this new constructor. I assume still a lot of work is required.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

RSE-dev, 2024-09-04:
I tested the new constructor.
Unfortunately, the result is the same (or maybe just a little different image of the snow)

Moreover, I have to add that the testing is running on two different platforms:
- Arduino mega2560, and its native IDE (the code execution gives only the black screen)
- STM32 with corresponding adaptations of the lib (the code execution gives a "snow" image)

For the future iterations, I'll run the code on both platforms to provide more information to you

olikraus, 2024-09-04:
Somehow it looks like that no data is transfered. 

- How does the code look like (for Arduino IDE)?
- Which is the exact constructor + arguments?
- How is the display connected? 
- For SPI: Can you confirm that the CD/A0 line is connected correctly?

RSE-dev, 2024-09-05:
I tested the code with two displays. The modules are absolutely the same except for the driver IC. The pinout of the modules is the same, so I'm just swapping the modules and flashing the new code to the Arduino Mega.

[image]

The code is pretty basic, please find it here with comments
`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

// U8G2_CH1120_128X160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);     //doesn't work with CH1120 based display
// U8G2_CH1120_128X160_2_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);     //doesn't work with CH1120 based display
// U8G2_CH1120_128X160_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);     //doesn't work with CH1120 based display

// U8G2_SH1108_128X160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);     //works fine with SH1108 based display
// U8G2_SH1108_128X160_2_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);     //works fine with SH1108 based display
U8G2_SH1108_128X160_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 42, /* dc=*/ 44, /* reset=*/ 46);    //works fine with SH1108 based display

char buf[16];

void setup(void) {
  u8g2.begin();  
}

void loop(void) {
  ltoa(millis(),buf,10);
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,24,"Hello World!");
    u8g2.drawStr(0,50,buf);
  } while ( u8g2.nextPage() );
  delay(1000);
}`

As the pinout of the modules is the same and the code works fine with the SH1108-based display, I assume, the connection is correct.

olikraus, 2024-09-07:
Ok, i found a remark in the datasheet and fixed the code based on this. Can you test the new beta (2.35.26) again ?

/
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

RSE-dev, 2024-09-09:
I tested the updated code. Unfortunately, there are no changes.
On Arduino Mega - just a black screen
On STM32 - a "snow" image that reacts to the screen content update by blinking its part

olikraus, 2024-09-09:
I am running out of ideas. Where did you buy that display?

RSE-dev, 2024-09-12:
I got it from TELEREX as a product sample. Suspecting the module is damaged, I ordered one more to check how it works. As soon as I have them received it, I will repeat the tests and let you know

RSE-dev, 2024-09-24:
I started the display after the modifying the initial sequence and the adding the U8X8_MSG_DISPLAY_DRAW_TILE functionality.
Again, there are some artifacts on the display
`#include "u8x8.h"

/* 
 code copyied from sh1107
 ch1120: 160x160 controller from Sino Wealth
 */

static const uint8_t u8x8_d_ch1120_noname_powersave0_seq[] = { U8X8_START_TRANSFER(), /* enable chip, delay is part of the transfer start */
U8X8_C(0x0af), /* display on */
U8X8_END_TRANSFER(), /* disable chip */
U8X8_END() /* end of sequence */
};

static const uint8_t u8x8_d_ch1120_noname_powersave1_seq[] = { U8X8_START_TRANSFER(), /* enable chip, delay is part of the transfer start */
U8X8_C(0x0ae), /* display off */
U8X8_END_TRANSFER(), /* disable chip */
U8X8_END() /* end of sequence */
};

static const uint8_t u8x8_d_ch1120_160x160_noname_powersave0_seq[] = { U8X8_START_TRANSFER(), /* enable chip, delay is part of the transfer start */
U8X8_C(0x0a0), /* segment remap a0/a1*/
U8X8_C(0x0c0), /* c0: scan dir normal, c8: reverse */
U8X8_END_TRANSFER(), /* disable chip */
U8X8_END() /* end of sequence */
};

static const uint8_t u8x8_d_ch1120_160x160_noname_powersave1_seq[] = { U8X8_START_TRANSFER(), /* enable chip, delay is part of the transfer start */
U8X8_C(0x0a1), /* segment remap a0/a1*/
U8X8_C(0x0c8), /* c0: scan dir normal, c8: reverse */
U8X8_END_TRANSFER(), /* disable chip */
U8X8_END() /* end of sequence */
};

static uint8_t u8x8_d_ch1120_generic(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
	uint8_t x, c, y;
	uint8_t *ptr;
	switch (msg) {
	/* handled by the calling function
	 case U8X8_MSG_DISPLAY_SETUP_MEMORY:
	 u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ch1120_64x128_noname_display_info);
	 break;
	 */
	/* handled by the calling function
	 case U8X8_MSG_DISPLAY_INIT:
	 u8x8_d_helper_display_init(u8x8);
	 u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_64x128_noname_init_seq);
	 break;
	 */
	case U8X8_MSG_DISPLAY_SET_POWER_SAVE:
		if (arg_int == 0)
			u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_noname_powersave0_seq);
		else
			u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_noname_powersave1_seq);
		break;
	case U8X8_MSG_DISPLAY_SET_FLIP_MODE:
		if (arg_int == 0) {
			u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_160x160_noname_powersave0_seq);
			u8x8->x_offset = u8x8->display_info->default_x_offset;
		} else {
			u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_160x160_noname_powersave1_seq);
			u8x8->x_offset = u8x8->display_info->flipmode_x_offset;
		}
		break;
#ifdef U8X8_WITH_SET_CONTRAST
	case U8X8_MSG_DISPLAY_SET_CONTRAST:
		u8x8_cad_StartTransfer(u8x8);
		u8x8_cad_SendCmd(u8x8, 0x081);
		u8x8_cad_SendArg(u8x8, arg_int); /* ch1120 has range from 0 to 255 */
[... 109 lines cut ...]
/* tile_height = */20,
/* default_x_offset = */0,
/* flipmode_x_offset = */0,
/* pixel_width = */128,
/* pixel_height = */160 };

uint8_t u8x8_d_ch1120_128x160(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {

	if (u8x8_d_ch1120_generic(u8x8, msg, arg_int, arg_ptr) != 0)
		return 1;

	switch (msg) {
	case U8X8_MSG_DISPLAY_INIT:
		u8x8_d_helper_display_init(u8x8);
		u8x8_cad_SendSequence(u8x8, u8x8_d_ch1120_128x160_noname_init_seq);
		break;
	case U8X8_MSG_DISPLAY_SETUP_MEMORY:
		u8x8_d_helper_display_setup_memory(u8x8, &u8x8_ch1120_128x160_noname_display_info);
		break;
	default:
		return u8x8_d_ch1120_generic(u8x8, msg, arg_int, arg_ptr);
	}
	return 1;
}`
[image]

Olstyle, 2025-01-15:
I can see in the wiki that the Chip is now listed as supported, 
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ch1120-128x160
but the results i get with the latest codes are exactly the ones posted here: "Snow" if I use sh1108 code, blank screen with ch1120 code.
So actually no advancement on this issue? @RSE-dev how did you proceed?

RSE-dev, 2025-01-15:
[u8x8_d_ch1120.zip](https://github.com/user-attachments/files/18426799/u8x8_d_ch1120.zip)
This should work. Try it please.
@olikraus hope you don't mind that I changed the file? If it works for @Olstyle please add the changes to the repo

Olstyle, 2025-01-16:
With the bit reordering code from the zip I get something like this (I use the plain C variant of the U8g2 code by the way):
[image]
The two small areas were supposed to be one MUI on the whole screen. It feels like this is not really mono mode.
As an exact display I have the WEO160128AWPP3N00000
As @RSE-dev  before, I got this as a sample to try out the supposedly "compatible updated OLED display with the SH1108 successor CH1120".
I have tried 0x0ac, 0x003 by the way: No change.

olikraus, 2025-01-16:
> "compatible updated OLED display with the SH1108 successor CH1120".

Looking at both datasheet, then I see a lot of differences. 

@Olstyle 
I can offer to do a kind of a remote development. I change the code and you could send me a picture of the resulting image.
I would first try to see, whether we can fix the problems by changing the configuration, but we might end up to takeover some code and ideas from @RSE-dev.

I would need to see how the following code looks like with the original unmodified current version of u8g2.
Not sure whether the code below is 100% correct and also the constructor  needs to be modified.

``` C++
U8G2_CH1120_128X160_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
void setup(void) {
  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.drawLine(0,0,100, 100);
    u8g2.setFont(u8g2_font_ncenB10_tr);
    u8g2.drawStr(0,30,"Hello World!");
  } while ( u8g2.nextPage() );
}
```

Please also send any modifications to the above code (along with the result on the display)... if you want me to code on this...

RSE-dev, 2025-01-17:
The code I provided works for me. Please pay attention, to the fact that the controller works at a limited frequency, try to slow down your bus. Also check the init section.
I apologize for the code formatting. I realize it's not perfect and may be not readable for everyone, but at least it's something that may people help and works in my environment.

Olstyle, 2025-01-17:
> The code I provided works for me. Please pay attention, to the fact that the controller works at a limited frequency, try to slow down your bus. Also check the init section. 

I tried with 2Mhz, 1MHz and 500kHz. All showed the same behavior. For now I am doing all tests at 500kHz.
 According to the datasheet 5Mhz (200ns), or even 20Mhz(50ns) if you take wait time between bytes into account, are allowed so I don't quite see why you think the chip doesn't like high frequencies.

>I apologize for the code formatting. I realize it's not perfect and may be not readable for everyone, but at least it's something that may people help and works in my environment.

Readability is no issue, it's just that without some more advanced merge tools like beyond compare it is hard to spot the actual changes because of the reformat.

> > "compatible updated OLED display with the SH1108 successor CH1120".
> 
> Looking at both datasheet, then I see a lot of differences.

I see that as well, this was only a quote of the supplier.
> 
> [@Olstyle](https://github.com/Olstyle) I can offer to do a kind of a remote development. I change the code and you could send me a picture of the resulting image. I would first try to see, whether we can fix the problems by changing the configuration, but we might end up to takeover some code and ideas from [@RSE-dev](https://github.com/RSE-dev).
> 
> I would need to see how the following code looks like with the original unmodified current version of u8g2. Not sure whether the code below is 100% correct and also the constructor needs to be modified.
> 
> ...
> Please also send any modifications to the above code (along with the result on the display)... if you want me to code on this...

Since I am on plain C I had to translate the Arduino code, so this is how it looks for me:

```c
u8g2_t DisplayControlStruct;

void displayu8g2_test()
{
    /* setup in one page mode */
    u8g2_Setup_ch1120_128x160_1(&DisplayControlStruct, U8G2_R0, displayu8g2_ByteCallback, displayu8g2_GpioDelayCallback);

    /* Init display and wake it up */
    u8g2_InitDisplay(&DisplayControlStruct);
    u8g2_SetPowerSave(&DisplayControlStruct, 0);
    u8g2_ClearDisplay(&DisplayControlStruct);

/* Allow for drawing */
    u8g2_FirstPage(&DisplayControlStruct);

    do{
        u8g2_DrawLine(&DisplayControlStruct, 0, 0, 100, 100);
        u8g2_SetFont(&DisplayControlStruct, u8g2_font_ncenB10_tr);
        u8g2_DrawStr(&DisplayControlStruct, 0, 30, "Hello World!");
    }while(u8g2_NextPage(&DisplayControlStruct));
}

```
The callbacks are my own which worked fine with sh1108
First thing I got was no image at all. By adding U8X8_C(0x0af), as the last command before U8X8_END_TRANSFER() in u8x8_d_ch1120_128x160_noname_init_seq[] I do get the "snow" effect again, but only after u8g2_SetPowerSave() which actually boils down to the same command again. 
The loop does not seem to change the displayed content.

olikraus, 2025-01-17:
Hmm... I somehow feel, that we need to fix more things here. And yes, the display seems to be in greylevel mode.

> By adding U8X8_C(0x0af), as the last command before U8X8_END_TRANSFER() 

This should not be required, because the only purpose of the u8g2_SetPowerSave() command is to send the 0xaf command.
Also the sendF command mentioned earlier should have put the display into monochrome mode, which actually didn't happen.
So before fixing the display, we probably need to fix your two callback functions.
Could you paste the callback functions displayu8g2_ByteCallback and displayu8g2_GpioDelayCallback here?

Olstyle, 2025-01-18:
> So before fixing the display, we probably need to fix your two callback functions.

I am with you there. Not all commands seem to be registered, which sounds like they are not send correctly. I'll whip out the oscilloscope and analyser on the lines when I'm back in the office.
Still surprising to me is the fact that those same callbacks work just fine for sh1108. So maybe the delay settings for cn1120 are not ok? Which might also explain why @RSE-dev is seeing better results by lowering the SPI frequency.

olikraus, 2025-01-18:
I can also review your code

olikraus, 2025-01-20:
> For some reason the first command sequence seems to get lost

One problem could be, that your code changes the DC (A0) signal level without waiting for the data transfer.
Ignore the DMA things on the code below, the same problem applies in cases, where you have a buffered transmit or a small FIFO buffer in the SPI subsystem: You need to wait until the SPI transfer is completly done. At least you need to wait until transfer completion before changing the DC or CS signal: See for example the "###" comment.

``` C++
uint8_t dma_buffer[256]; /* required for DMA transfer */

extern "C" uint8_t u8x8_byte_dma_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) {
  uint16_t i;
  switch(msg) {
    case U8X8_MSG_BYTE_SEND:
      /* wait for DMA completion */
      /* ... */      
      /* create a copy of the input data */
      for( i = 0; i < arg_int; i++ )
        dma_buffer[i] = ((uint8_t *)arg_ptr)[i];
      /* create DMA SPI Transfer for arg_int bytes, located at dma_buffer */
      /* ... */
      break;
    case U8X8_MSG_BYTE_INIT:
      /* setup SPI & DMA */
      /* ... */
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ### Wait for DMA SPI transfer completion ### */
      /* ... */
      u8x8_gpio_SetDC(u8x8, arg_int);
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);  
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:      
      /* ### Wait for DMA SPI transfer completion ### */
      /* ... */
      u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
      u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
      break;
    default:
      return 0;
  }  
  return 1;
}
```

Olstyle, 2025-01-29:
> > For some reason the first command sequence seems to get lost
> 
> One problem could be, that your code changes the DC (A0) signal level without waiting for the data transfer. Ignore the DMA things on the code below, the same problem applies in cases, where you have a buffered transmit or a small FIFO buffer in the SPI subsystem: You need to wait until the SPI transfer is completely done. At least you need to wait until transfer completion before changing the DC or CS signal

Sorry for the late response. Only today I managed to do some measurements.
This is `u8g2_InitDisplay` in the configuration which works for me with one command and CS in between
[image]
This is the start of `u8g2_ClearDisplay`
[image]
This is the end of one block in `u8g2_ClearDisplay`
[image]
(yellow is CLK, blue CS and red DC(A0))
As `u8g2_ClearDisplay` should do addressing and then data, and I can find no CS or DC changes during active clock, this looks correct to me.

olikraus, 2025-01-29:
Nice pictures, but not sure exactly this means. At least it looks to me that there is a lot of cross talk between lines.
Anyway, is your problem solved?

Olstyle, 2025-01-30:
> Nice pictures, but not sure exactly this means. At least it looks to me that there is a lot of cross talk between lines. Anyway, is your problem solved?

The crosstalk is mainly bad grounding during the measurement, nothing to worry.
What you see is that I had an SPI issue which is now solved. I'm now problem free using the code from @RSE-dev . 
If there is something you still want to have tested with your original code I could now do it.

olikraus, 2025-02-06:
ok, i don't have much time at the moment, so I would keep it as it is for now...