olikraus/u8g2 #2504: I dont't know what constructor use.

JPRIPKE, 2024-09-09:
I did a program with the u8g2 lib and at the wokwi simulator worked very well, but I was trying to use in the real life and don't work.
Probably the error is the constructor, i have tried all the constructor of the doc.
Can anyone help me? or explain, how can i discover what constructor i need to use? 

Informations:
SSD1309
128x64
I2C
2,42"

olikraus, 2024-09-09:
Probably you need to use one of the SSD1309 constructors:
https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#ssd1309-128x64_noname0

JPRIPKE, 2024-09-10:
Yes, i tried all of that list, but none worked, i wanna to know how can i choose the correct one.
I must try again that constructors.
tr Thanks.

JPRIPKE, 2024-09-10:
U8GLIB_SSD1309_128X64 u8g(U8G_I2C_OPT_DEV_0 | U8G_I2C_OPT_NO_ACK);

When i use the u8g Lib that constructor works very well, probably that information could be important

olikraus, 2024-09-10:
I suggest to try with "U8G2_SSD1309_128X64_NONAME0_F_SW_I2C" or "U8G2_SSD1309_128X64_NONAME2_F_SW_I2C". Provide the pin numbers where your display is connected.
This constructor should be compatible with the u8g constructor.

JPRIPKE, 2024-09-10:
nonething happened, read the code, maybe you can see any wrong 
I'm using the arduino mega, whithout change the pinout of SDA and SCLK

U8G2_SSD1309_128X64_NONAME0_F_SW_I2C u8g2(/*rotation*/U8G2_R0,/*clock*/ 21,/*data*/ 20, /*reset*/ U8X8_PIN_NONE);

//U8G2_SSD1309_128X64_NONAME2_F_SW_I2C u8g2(/*rotation*/U8G2_R0,/*clock*/ 21,/*data*/ 20, /*reset*/ U8X8_PIN_NONE);

JPRIPKE, 2024-09-10:
Worked, thank you. I tried a example code, probably i wrote anything wrong at my code.
Do you have any tip to me debug? Because when i run that code at wokwi run well

JPRIPKE, 2024-09-10:
i just discovered the problem, when i write "rtc.begin();" the display stop the work.
i going to reserch to fix that.

JPRIPKE, 2024-09-10:
"SSD1306 display supports I2C at 400KHz, the DS1307 does not. if the RTC library is poorly written and does not check the bus frequency before using it as per arduino guidance then it will fail to work. you should be able to either set the clock to 100KHz yourself prior to using the RTC, or set the bus clock to 100KHz in u8g2" I found that, could you explain to me where do that change into the (.h) or (.cpp) of u8g2 lib

olikraus, 2024-09-10:
You can slow down u8g2 as mentioned in https://github.com/olikraus/u8g2/issues/1392