olikraus/u8g2 #2507: broken font links... no hebrew fonts available.

giggty, 2024-09-14:
broken links for hebrew font's : 
u8g2_font_cu12_t_hebrew [cu12](https://github.com/olikraus/u8g2/wiki/fntgrpcu12)
X11 font sources are hosted [here](http://cgit.freedesktop.org/xorg/font/).
Font download location is [here](http://sofia.nmsu.edu/~mleisher/Software/cu/).

are there any hebrew fonts available?

olikraus, 2024-09-15:
What exactly do you mean by "broken link"?
Pictures are visible and the fonts are available, you can just use them:

u8g2_font_6x13_t_hebrew
u8g2_font_6x13B_t_hebrew
u8g2_font_cu12_t_hebrew
u8g2_font_unifont_t_hebrew

Of course additional u8g2 fonts could be generated from any bdf fonts...

giggty, 2024-09-15:
sorry ... i had the wrong impression that you had to download the font .
trying to figure out how to change size since there is no setfont size or somthing like that ... ?

olikraus, 2024-09-17:
> trying to figure out how to change size since there is no setfont size or somthing like that ... ?

Indeed, bitmap fonts can't be scaled on the fly to a different size.