olikraus/u8g2 #2508: Unable to intercept or exit the menu [MUI]

martino87r, 2024-09-16:
Unless I'm missing something, the default function for exiting the menu is wired incorrectly to the events:
https://github.com/olikraus/u8g2/blob/8cfcc70c35c9c82f6e4723b926401e17053a4152/csrc/mui_u8g2.c#L873

Whenever i want the menu to exit, it does nothing.

I'm really bad at c/cpp so forgive me if this is totally unfounded, just implemented the default example with 1 exit button and wanted to intercept the action in order to set my own state to trigger a repaint of my own UI...

```c
muif_t muif_list[] = {  
  MUIF_U8G2_FONT_STYLE(0, u8g2_font_helvR08_tr),  /* define style 0 */
  MUIF_U8G2_LABEL(),                                                    /* allow MUI_LABEL command */
  MUIF_BUTTON("BN", exit_menu)        /* define exit button */
};

fds_t fds_data[] =                              /* Don't use comma between the commands! */
MUI_FORM(1)                                     /* This will start the definition of form 1 */
MUI_STYLE(0)                                    /* select the font defined with style 0 */
MUI_LABEL(5, 15, "Hello U8g2")          /* place text at postion x=5, y=15 */
MUI_XYT("BN",64, 30, " Exit ")     /* place the exit button at pos x=64, y=30 */
;

```

```c
// MUI functions call
uint8_t exit_menu(mui_t *ui, uint8_t msg){
  // Forward handling to real function
  // We intercept only the interesting bit here
  if (msg == MUIF_MSG_CURSOR_SELECT) {
    SerialUSB.println("Exiting menu");
    // Set my own vars here to signal the menu is closing
  }
  if (msg == MUIF_MSG_VALUE_DECREMENT) {
    SerialUSB.println("Decrement?");
  }
  return mui_u8g2_btn_exit_wm_fi(ui, msg);
}
```

```
01:47:21.340 -> Long pressed SELECT button
01:47:21.340 -> Redrawing mui
01:47:24.547 -> Pressed SELECT button
01:47:24.547 -> Exiting menu
01:47:32.502 -> Pressed DOWN button
01:47:34.204 -> Pressed UP button
```

Of course decrement never triggers since I don't expect to trigger a button by decrementing it...

olikraus, 2024-09-19:
Ok, this is a good question. The increment and decrement doesn't make sense for the exit button, but i probably found it clever to exit the menu also with the increment / decrement events.
There is an example which uses the exit button:
https://github.com/olikraus/u8g2/blob/416abe4af4e5afd10045c42e46c0d844b6761ddc/sys/arduino/u8g2_page_buffer/MUIMinimal/MUIMinimal.ino#L390-L404

martino87r, 2024-09-24:
Thanks for the info, I managed to fix my issue by simply ignoring the internal wiring on the menu and simply intercept the proper event, bubble it up and finally perform my work.

Ex.

```
// MUI functions call
uint8_t exit_menu(mui_t *ui, uint8_t msg){
  // Forward handling to real function
  // We intercept only the interesting bit here
  if (msg == MUIF_MSG_CURSOR_SELECT) {
    SerialUSB.println("Exiting menu");
    // Set my own vars here to signal the menu is closing
    u8g2.clearDisplay();
    // Force a reload of the main UI
    needs_redraw = true;
  }
  return mui_u8g2_btn_exit_wm_fi(ui, msg);
}
```

closed 2024-09-24 by martino87r