olikraus/u8g2 #2518: Support for S1d13700 240x64

DinoxGaming, 2024-10-05:
I am doing a project where i need to get a Batron BT240064 working with arduino. This is a 240x64 dots lcd. It uses the Epson S1D13700 lcd controller (SED1330 replacement). I tried to use the u8g2 SED1330 constructor. "U8G2_SED1330_240X128_1_8080 u8g2(U8G2_R0, 2, 3, 4, 5, 6, 7, 8, 9, /*enable=*/ 10, /*cs=*/ 12, /*dc=*/ 13, /*reset=*/ A4); " This is how i connected it to my arduino uno and I connected the read to 5v and left the contrast pins unattached as it says in the datasheet. I took tis part from an old control panel and when i use the original control panel pcb, it appears that the lcd is working because there appears a darker background on the pixels after initliazing. but when i try to do this with u8g2 nothing happens. I do not see a darker background appear and i do not see any pixels. I am using the Graphicstest example code. I attached pictures what i see with the lcd connected to arduino and i attached a picture of it attached to the original pcb. what i was expecting to also see with u8g2.
[image]
[image]
[image]

olikraus, 2024-10-05:
Please check the wiring first. Maybe post a table here which pin of the display is connected to which pin on your Arduino. 

I don't understand the datasheet why not to connect v0. From my end, V0 needs to be connected to the wiper of a varpot for proper adjustment of the contrast (other ends usually go to vee and gnd).

DinoxGaming, 2024-10-05:
@olikraus 
The contrast is set by a soldered jumper on the backside of the lcd. The contrast has a fixed value set by the pcb of the lcd module. 

My connections from lcd to arduino:
D0 to 2
D1 to 3
D2 to 4
D3 to 5
D4 to 6
D5 to 7
D6 to 8
D7 to 9

RS Register Select to 13
CS Chip select to 12
WR Write to 10
RD Read to 5v
RES Reset to A4.

In the s1d13700 datasheet it says that after setting the system set command the YDIS pin on the controller turns high and i tested this with an LED. After sending the u8g2.begin(); this Pin indeed turns high. So that means that the lcd is correctly responding to the data sent by u8g2. But no signals are sent to the drivers. But when i connect the lcb to the original pcb i see that the lcd works because i see the darker background on the pixels as seen in the image i added earlier. And then the lcd srivers

olikraus, 2024-10-05:
I think this fits to your constructor:
``` C++
U8G2_SED1330_240X128_1_8080 u8g2(U8G2_R0, 2, 3, 4, 5, 6, 7, 8, 9, /*enable=*/ 10, /*cs=*/ 12, /*dc=*/ 13, /*reset=*/ A4); 
```
However the 240x128 is wrong for your display and would also explain the wrong contrast above.
At least we need to adjust the display configuration for your display. After u8g2.begin() you should setup the display for 240x64 pixel. 
The setup sequence looks like this: 
[image]

I wild guess could be this:
``` C++
u8g2.sendF("caaaaaaaa", 0x40, 0x30, 0x87, 0x07, 0x1d, 0x50, /*L/F*/ 0x40, 0x1e, 0x00);
```

I think L/F is the display height, so it should be 0x40 or 0x3f (64 or 63). Other values I am not sure. Maybe you can look into the controller manual and play around with the values. Once you figured out the correct setup, I can also create a 240x64 constructor for this display.

DinoxGaming, 2024-10-05:
@olikraus The command you provided managed to get the display fully working. So this could be added as a new constructor. i needed to change L/F to 3F.

olikraus, 2024-10-06:
Nice.

The sendF command is described her: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf
If you want to send one command with one arg, then you need to use "ca" as first argument for sendF.
The signal levels for CS, DC (=A0) and WR are described in the manual, but for the setup command it is also given in the last 3 columns of the screenshot above, so for 

``` C++
u8g2.sendF("caaaaaaaa", 0x40, 0x30, 0x87, 0x07, 0x1d, 0x50, /*L/F*/ 0x3f, 0x1e, 0x00);
```

the value 0x40 is the command and the DC signal (= A0 signal) has to be one. For all other args the DC signal has to be low. CS will be low for the complete sequence and will be set to 1 once all bytes are transmitted.

DinoxGaming, 2024-10-06:
> Nice.
> 
> The sendF command is described here: https://github.com/olikraus/u8g2/wiki/u8g2reference#sendf If you want to send one command with one arg, then you need to use "ca" as first argument for sendF. The signal levels for CS, DC (=A0) and WR are described in the controller datasheet, but for the setup command it is also given in the last 3 columns of the screenshot above, so for
> 
> ```c++
> u8g2.sendF("caaaaaaaa", 0x40, 0x30, 0x87, 0x07, 0x1d, 0x50, /*L/F*/ 0x3f, 0x1e, 0x00);
> ```
> 
> the value 0x40 is the command and the DC signal (= A0 signal) has to be one (high). For all other args the DC signal has to be low. CS will be low for the complete sequence and will be set to 1 once all bytes are transmitted.

DinoxGaming, 2024-10-06:
Thank you so much for your quick responses and your help!

olikraus, 2024-10-09:
reopen for constructor implementation

olikraus, 2024-10-09:
New Constructor: U8G2_SED1330_240X64_1_8080

olikraus, 2024-10-09:
I have created a new beta release 2.36.1, maybe you can test the above constructor...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

DinoxGaming, 2024-10-10:
@olikraus Using the 240x64 constructor and then using u8g2.begin() does not initilialize my lcd. However when i use the constructor and after u8g2.begin i send the u8g2.sendF("caaaaaaaa", 0x40, 0x30, 0x87, 0x07, 0x1d, 0x50, /*L/F*/ 0x3f, 0x1e, 0x00); then it does initialize correctly. But not when using the 240x64 constructor.

olikraus, 2024-10-11:
Thanks for the feedback, I will recheck this.

olikraus, 2024-10-12:
Everything looks ok. The correct value 0x3f is available: https://github.com/olikraus/U8g2_Arduino/blob/485ad93cc983a929d4a5b6333f4bc3a655f827f2/src/clib/u8x8_d_sed1330.c#L300-L308

and the correct init sequence is also called:
https://github.com/olikraus/U8g2_Arduino/blob/485ad93cc983a929d4a5b6333f4bc3a655f827f2/src/clib/u8x8_d_sed1330.c#L342

Are you sure that you had used the new constructor?

DinoxGaming, 2024-10-12:
@olikraus Yes i used the new constructor. But it only works wben sending. u8g2.sendF("caaaaaaaa", 0x40, 0x30, 0x87, 0x07, 0x1d, 0x50, /L/F/ 0x3f, 0x1e, 0x00) after u8g2.begin(). I tried testing random values here and the display still initialized. It did not display the graphics correctly but it did still initialized. But not when using only u8g2.begin() with the 240x64 constructor. Maybe it has something to do with the order u8g2.begin sends the commands/parameters?

olikraus, 2024-10-13:
hmm i have added the same command again. So indeed the init will be now sent twice:

https://github.com/olikraus/u8g2/blob/9c841fafef11f430a225aa692944264c26a2405d/csrc/u8x8_d_sed1330.c#L343

With this, i created another beta 2.36.2

If this still doesn't work, can you send your test code here?

Thanks....

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

DinoxGaming, 2024-10-16:
@olikraus I accidently destroyed the oscillator  circuit on the PCB of this lcd. So the LCD Module does not work anymore. I do not think that this is repairable I can not test the code anymore.

olikraus, 2024-10-16:
oh, too bad... 
Thanks for the feedback.

closed 2024-10-16 by olikraus