olikraus/u8g2 #2520: u8g2 repository unusable

carramar, 2024-10-09:
This issue is not with the library but with the actual github page for the fonts. >50% of the font images are broken or missing. and of the font images that actually do load, 90% of them are wrong. in some cases you'll have 5 identical images in a row for 5 fonts that look nothing alike.

This horrendous listing page is so bad the best way to find a font is to just go down the list reprogramming the microcontroller for every single font until one of them looks about right. he general attitude toward this library is "oh god not u8g2" and i suspect 95% of that sentiment is due to the font list page.

please for the love of god and github, fix it.

olikraus, 2024-10-09:
Which page / URL did you use? I already did a lot to prevent load issues.. From here:
https://github.com/olikraus/u8g2/wiki/fntlistall
Best is to use  https://github.com/olikraus/u8g2/wiki/fntgrp

I could also offer to create pdf versions of the fonts pages...

olikraus, 2024-10-09:
BTW: For me everything works without problem.