olikraus/u8g2 #2522: SSD1306 Display is not working with Bluepill Board STM32F103

Jaishankar872, 2024-10-13:
### Hardware Details
MCU -  Bluepill  STM32F103CB 
OLED  SSD1306 128x32 I2C display
 PA3 - SDA
 PA4 - SCL
[image]

### Code (in PlatformIO)
  First I used the softwire library's scanner code to validity I2C connection of PA3(SDA) & PA4 (SCL). Got the Slave address as 0x3C 
  With the below code Display is not working(showing nothing)
```
#include <U8g2lib.h>
U8G2_SSD1306_128X32_UNIVISION_F_SW_I2C display(U8G2_R0, PA4, PA3, U8X8_PIN_NONE);
void setup()
{
display.begin();
}
void loop()
{
  display.clearBuffer();
  display.setFont(u8g2_font_ncenB08_tr);
  display.drawStr(0, 10, "Hello STM32!");
  display.sendBuffer();
  delay(1000);
}
```
### Digital analyzer waveform output
[image]

## Problem Statement

After seeing the code and Logic analyzer waveform, there are some issue configuration in the code while use U8g2lib library. Please help here solve this issue

Thanks in Advance!!!

olikraus, 2024-10-13:
Difficult.
So far i personally have only used Arduino IDE. Moreover I how only "official" boards. So I can't help much here. In fact I have no idea regarding this issue...

Jaishankar872, 2024-10-18:
@olikraus 
When I tried the ACROBOTIC_SSD1306 library display working fine. Only issue is rising while using this U8g2 library.

Refer to the post for more details & code Used.
https://community.platformio.org/t/ssd1306-display-is-not-working-on-bluepill-board-stm32f103/43752/6?u=jaishankar

maxgerhardt, 2024-10-18:
This seems to be an issue with the software-i2c implementation on STM32 in this library. It should also reproduce on a regular Bluepill board, or any official STM32 board.