olikraus/u8g2 #2523: Unused function and unused variable

tbnobody, 2024-10-14:
When compiling with higher warning levels (`-Wall -Wextra -Wunused`) I receive the following warn message (or error message when compiling with `-Werror`
```
Compiling .pio\build\generic_esp32s3_usb\lib55e\U8g2\clib\u8g2_buffer.c.o
.pio/libdeps/generic_esp32s3_usb/U8g2/src/clib/mui.c:591:16: warning: 'mui_task_read_nth_selectable_field' defined but not used [-Wunused-function]
 static uint8_t mui_task_read_nth_selectable_field(mui_t *ui)
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\generic_esp32s3_usb\lib55e\U8g2\clib\u8g2_button.c.o
Compiling .pio\build\generic_esp32s3_usb\lib55e\U8g2\clib\u8g2_circle.c.o
.pio/libdeps/generic_esp32s3_usb/U8g2/src/clib/mui_u8g2.c: In function 'mui_u8g2_s8_vmm_draw_wm_pf':
.pio/libdeps/generic_esp32s3_usb/U8g2/src/clib/mui_u8g2.c:1256:9: warning: unused variable 's' [-Wunused-variable]
   char *s = buf;
         ^
```

olikraus, 2024-10-14:
Thanks, assigned to next release

olikraus, 2025-03-12:
I will look into this. Thanks for reporting.