olikraus/u8g2 #2526: Loading font data from external storage,  such as TF card or SPIFLASH

meibao-real, 2024-10-20:
Hello
Firstly, u8g2 is a fantastic project.
Sometimes we need to use custom font data, and using bdfconv can create custom font data, but!! when I have a lot of fonts data and the MCU's Flash capacity is small, so there's a problem. such as I ported u8g2 to Stm32f103c8t6, the Stm32f103c8t6 flash has 64kb, but the fonts data has 200kb or more.....
So can that package the font data into Bin binary(or other?) and store it in external memory, such as TF card or SPIFLASH, and then read the binary data for font loading when needed
such as ..
`
uint8_t* diyfont = fontsloadformsd("xxxx.bin") // form tf card load some font data
u8g2.SetFont(diyfont)
`
so , What should I do if possible ?
thanks.

meibao-real, 2024-10-20:
Um ....
It seems unlikely because even if these data are loaded from external storage, the read font data will consume the corresponding RAM, such as reading a 100KB file from the TF card, but the MCU's internal RAM is only 20kb.......

oh ...

olikraus, 2024-10-20:
see also here: https://github.com/olikraus/u8g2/issues/461

meibao-real, 2024-10-21:
> see also here: #461

Ohh thanks.
but .. I have found another solution, such as using a font chip.
For example, GT22L16A1Y is a dot matrix font chip that includes Chinese, Japanese, and Korean. I am planning to port it to U8g2 to solve the problem. If there are no issues with the verification, I will submit a PR to enable U8g2 to support external font chips for displaying text.
As mentioned at the beginning, u8g2 is a great project that is being used by people from many countries. Some countries do not use Latin letters as their characters and need to display their own country's characters, such as China(中文), Japan(にっぽんこく), Korea(대한민국), or more(????)....
So creating corresponding font files will consume a lot of valuable Flash storage space that is already limited.
I hope using an external font chip can solve my problem.
When I solve the problem, I will close this issue~~~~~

olikraus, 2024-10-29:
sure, however i personally would assume that using a uC with more flash memory should still be easier.