olikraus/u8g2 #2527: convert Arabic fonts( Unicode UTF-8 ) not work !!

X960xx, 2024-10-26:
hello olikraus

I tried many times to convert Arabic fonts to work in my project, but I did not succeed. 
I do not know what the reason is ?

from ttf to BDF
otf2bdf -r 72 -p 24 segoeui.ttf -o segoeui.bdf

from BDF to C file
bdfconv -f 1 -m '32-255' -n segoeui-o segoeui.c segoeui.bdf
bdfconv -v 0 -f 1 -m'32-128,$600-$06FF,$FB50-$FBB1,$FE70-$FEFF' -n u8g2_font_segoeui -o u8g2_font_segoeui.c segoeui.bdf
bdfconv -b 0 -f 1 -m map/gb2312.map -n u8g2_font_segoeui -o u8g2_font_segoeui.c segoeui.bdf

Any help ?

thank you

olikraus, 2024-10-29:
Which command doesn't work?
Basically you need a blank between the option and its arguments. Also the map file requires an uppercase -M

bdfconv [options] filename
-h          Display this help
-v          Print log messages
-b <n>      Font build mode, 0: proportional, 1: common height, 2: monospace, 3: multiple of 8, 4: 5x7 mode
-f <n>      Font format, 0: ucglib font, 1: u8g2 font, 2: u8g2 uncompressed 8x8 font (enforces -b 3)
-m 'map'    Unicode ASCII mapping
-M 'mapfile'    Read Unicode ASCII mapping from file 'mapname'
-o <file>   C output font file
-k <file>   C output file with kerning information
-p <%>      Minimum distance for kerning in percent of the global char width (lower values: Smaller gaps, more data)
-x <n>      X-Offset for 8x8 font sub-glyph extraction (requires -f 2, default 0)
-y <n>      Y-Offset for 8x8 font sub-glyph extraction (requires -f 2, default 0)
-th <n>     Horizontal size of the 8x8 glyphs (requires -f 2, default 1)
-tv <n>     Vertical size of the 8x8 glyphs (requires -f 2, default 1)
-n <name>   C indentifier (font name)
-d <file>   Overview picture: Enable generation of bdf.tga and assign BDF font <file> for description
-l <margin> Overview picture: Set left margin
-g <glyphs> Overview picture: Set glyphs per line (default: 16)
-a          Overview picture: Additional font information (background, orange&blue dot)
-t          Overview picture: Test string (Woven silk pyjamas exchanged for blue quartz.)
-r          Runtime test

X960xx, 2024-10-30:
HELLO OLIVER

ALL commandS IN THIS SITE NOT WORK WITH ME

I USEd command FROM THIS WEBSITE u8g2 Unifont helper
https://stncrn.github.io/u8g2-unifont-helper
 I USED THIS commands AND IT'S WORKING

TTF > BDF
otf2bdf -r 72 -p 32 Hacen.ttf -o myfont.bdf

BDF > C FILE
bdfconv -v -f 1 -m "32-65276" myfont.bdf -o u8g2_font_myfont.c -n u8g2_font_myfont -d myfont.bdf

I noticed most Arabic fonts maps start from 32 and end at 65276

thank you

olikraus, 2024-11-03:
I am happy to see that your problem was solved with the help of an external web site.