olikraus/u8g2 #2528: ssd1306 NO_NAME I2C FW not work rp2040

tistructor, 2024-11-04:
Hi,
I'm testing with sdd1306 in I2C on a board with rp2040.

It works well if I use I2C in HW but if I use software mode the display remains black.

I used the sample file attached to the library, especially the graphicTest.ino.

I also tried to change the gpio but I don't get results.

tistructor, 2024-11-05:
I checked all the problems with rp2040 but there is nothing about the fact that I2C software doesn't work.
I tried in every possible way but the display remains off.
I set the same I2C HW GPIOs in software mode but nothing, it doesn't work.
I'm using the https://github.com/earlephilhower/arduino-pico package
latest version.
I tried adding external pullup resistors on the scl and sda pins but the problem remains.
I can check the quality of the signal and whether it is present with an oscilloscope.

I'm using micro defaul settings no overclocking or anything.

tistructor, 2024-11-05:
I could try powering the display with 5V as it accepts this power while keeping the data level at 3.3V.

But first I'll do a test with 5V arduino and software mode and see if everything works.

tistructor, 2024-11-05:
I tried changing gpio for i2c software and the scan results in the same 2 addresses.

tistructor, 2024-11-05:
However, the peripheral in I2C sw is detected.
I think it depends on some timing within your library.
I couldn't say anything else.

tistructor, 2024-11-05:
Software mode that doesn't work:
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, /* clock=*/ D8, /* data=*/ D7, /* reset=*/ U8X8_PIN_NONE);

Hardware mode that works:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

tistructor, 2024-11-05:
I also did the same graphicTest on an rp2040 module and it works the same way as my coustom card.
I2C HW works and FW mode doesn't work.

tistructor, 2024-11-05:
Tested on arduino one, I2C works both fw and hw.

I also scanned I2C addresses with softwire and in this case I get the correct address 0x3C.

I think there are timing problems with RP2040 in software mode.

olikraus, 2024-11-05:
Thanks for the testing and reporting this here. I agree to you conclusion and i am sorry not to be able to help here...