olikraus/u8g2 #2529: use and edit 16 bit numbers  in MUI menu system.

czorgormez, 2024-11-05:
is it possible to use and edit 16 bit numbers in mui menu system ?

olikraus, 2024-11-05:
I think a 16 bit var doesn't make much sense. How shell the user enter 10000? The user needs to click 10000 times to reach that value. I think it is better to break down the number into 5 variables, each from 0 to 9 (first var from 0 to 5). Then you can calculate the 16 bit value with a*10000+b*1000+c*100*d*10+e (a..e are the five variables).
 
There is a similar example here:
https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/MUIFractionSimpleRotary/MUIFractionSimpleRotary.ino

The above example also updates the next higher value based on the overflow of the lower number however this is not really required.

czorgormez, 2024-11-05:
agree, it is not practical to edit numbers larger than 100.

however we used to edit this type of variables with cursor select system thousandth, hundredth, tenth, oneth  one by one and user can select the digit and edit by up-down buttons.

also there is another old method called as typematic. a single click adds or subtracts one but if user keeps pressing the button variable starts to increment tenth every second, after several sec. it starts to increment by hundreds every sec etc.

olikraus, 2024-11-05:
true, but this could be simply implemented by yourself: Just create a new muif, which displays your 16 bit variable:
```
uint8_t muif_draw_16_bit_variable(mui_t *ui, uint8_t msg) {
  if ( msg == MUIF_MSG_DRAW ) {
      u8g2.setCursor(mui_get_x(ui), mui_get_y(ui));
      u8g2.print(my_16bit_var);
  }
  return 0;
}
```
Register this muif in the muif table (for example with "VV")and refer the same inside FDS with MUI_XY("VV",x,y).

Then, if the form with this number is visible (by checking the FDS form id with https://github.com/olikraus/u8g2/wiki/muiref#getcurrentformid) , just apply your up / down updates and do a redraw of the menu.

czorgormez, 2024-11-05:
Thanks, i will use this method.