olikraus/u8g2 #2531: Display a bitmap with the DrawTile function...

FredM67, 2024-11-08:
I've a bitmap of 72x64, which I can display without problem in "buffer mode".
Since this method needs too much ram, I wanted to draw this bitmap with the drawTile function.
I've tried to cut the bitmap in any possible way, I just get "garbage" on the display.
Here's the code:
```cpp
const unsigned char logo_bmp[] PROGMEM = {/*DATA*/};
constexpr uint8_t TILE_WIDTH{8};
constexpr uint8_t TILE_HEIGHT{8};
void drawTile(U8X8_SSD1306_128X64_NONAME_HW_I2C &u8x8, const uint8_t *bitmap, uint8_t tileX, uint8_t tileY)
{
  uint8_t tile[TILE_HEIGHT];
  for (uint8_t i = 0; i < TILE_HEIGHT; i++)
  {
   tile[i] = pgm_read_byte(bitmap + (tileY * TILE_HEIGHT + i) * (72 / 8) + tileX);
  }
  u8x8.drawTile(tileX, tileY, 1, tile);
}
```
and then in **setup**:
```cpp
  // Loop through the bitmap and draw each tile
  for (uint8_t y = 0; y < (64 / TILE_HEIGHT); y++)
  {
    for (uint8_t x = 0; x < (72 / TILE_WIDTH); x++)
    {
      drawTile(u8x8, logo_bmp, x, y);
    }
  }
```

Could someone tell me what I'm doing wrong ?
Or is it simply impossible to do that ?!?
Or is the **drawTile** method broken ?

Thx
Fred

olikraus, 2024-11-08:
I think you need a bitlevel converion of the 8x8 bitmatrix.
A tile (usually) is a bit matrix which looks like this: https://github.com/olikraus/u8g2/wiki/u8x8reference#drawtile

Whereas a BMP has a mirrored bit structure:
byte 0: bit 7, bit 6, bit 5..., bit 0
byte 1: bit 7, bit 6, ...

ok I even dont know if the the BMP starts with MSB or LSB... needs to be figured out...

olikraus, 2024-11-10:
I think that online tool is not able to swap the bytes...

olikraus, 2024-11-10:
I have created an example:
https://github.com/olikraus/u8g2/blob/d462b189e1795e8244488cdc4daa061d1918bd82/sys/sdl/u8x8_xbm/main.c#L27-L67

You probably want to use line 64 instead of line 63. 
Moreover this procedure works for XBM, so best is to take your image and save the image as .xbm with "gimp" tool.
Limitation:
 * not speed optimized
 * image width and height must be multiple of 8

FredM67, 2024-11-10:
Thx a lot, I'll try it out tomorrow.

olikraus, 2024-11-11:
I have created a complete Arduino example, which also puts the image into PROGMEM area: https://github.com/olikraus/u8g2/blob/b88a767cdfff3fc5c5029ba8b5b2b1470ef60400/sys/arduino/u8x8/DrawXBM/DrawXBM.ino

FredM67, 2024-11-11:
A huge thank you for your help, the new example...
I works like a charm ;-)