olikraus/u8g2 #2532: U8g2_for_Adafruit_GFX conflicts with U8g2

glenlau228, 2024-11-10:
1909 | extern const uint8_t u8g2_font_px437wyse700a_mn[] U8G2_FONT_SECTION("u8g2_font_px437wyse700a_mn");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3961:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_tf []' with 'C' linkage
 3961 | extern const uint8_t u8g2_font_px437wyse700b_tf[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tf");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1910:22: note: previous declaration with 'C++' linkage
 1910 | extern const uint8_t u8g2_font_px437wyse700b_tf[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tf");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3962:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_tr []' with 'C' linkage
 3962 | extern const uint8_t u8g2_font_px437wyse700b_tr[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tr");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1911:22: note: previous declaration with 'C++' linkage
 1911 | extern const uint8_t u8g2_font_px437wyse700b_tr[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tr");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3963:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_tn []' with 'C' linkage
 3963 | extern const uint8_t u8g2_font_px437wyse700b_tn[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tn");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1912:22: note: previous declaration with 'C++' linkage
 1912 | extern const uint8_t u8g2_font_px437wyse700b_tn[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_tn");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3964:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_mf []' with 'C' linkage
 3964 | extern const uint8_t u8g2_font_px437wyse700b_mf[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mf");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1913:22: note: previous declaration with 'C++' linkage
 1913 | extern const uint8_t u8g2_font_px437wyse700b_mf[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mf");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3965:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_mr []' with 'C' linkage
 3965 | extern const uint8_t u8g2_font_px437wyse700b_mr[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mr");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1914:22: note: previous declaration with 'C++' linkage
 1914 | extern const uint8_t u8g2_font_px437wyse700b_mr[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mr");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2\src/clib/u8g2.h:3966:22: error: conflicting declaration of 'const uint8_t u8g2_font_px437wyse700b_mn []' with 'C' linkage
 3966 | extern const uint8_t u8g2_font_px437wyse700b_mn[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mn");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~
d:\Arduino\libraries\U8g2_for_Adafruit_GFX\src/U8g2_for_Adafruit_GFX.h:1915:22: note: previous declaration with 'C++' linkage
 1915 | extern const uint8_t u8g2_font_px437wyse700b_mn[] U8G2_FONT_SECTION("u8g2_font_px437wyse700b_mn");
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~

olikraus, 2024-11-10:
yes, but why would you want to use these two libs together?

glenlau228, 2024-11-11:
Because I want to drive an ink screen and an LED screen at the same time, the ink screen uses the GxEPD2 driver, and I need to use U8g2_for_Adafruit_GFX to customize the font, and the LED screen needs to use the U8g2 driver. Is there any other way?

olikraus, 2024-11-11:
hmm It should be possible to just remove the font declarations from U8g2_for_Adafruit_GFX .h file, because fonts are exactly the same in U8g2_for_Adafruit_GFX and u8g2