olikraus/u8g2 #2533: MUI_GOTO isn't show on form

marhalloweenvt, 2024-11-13:
Hello,
Today I am trying some sample from MUI wiki and facing a very weird issue. It seems that MUI_GOTO will not be rendered on my form.
Below is my code:

`
U8G2_SSD1306_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE, I2C_SCL, I2C_SDA);
MUIU8G2 mui;

uint8_t menu_static_part(mui_t *ui, uint8_t msg) {
    switch (msg) {
    case MUIF_MSG_DRAW:
        u8g2.setFont(u8g2_font_ncenB08_tr);
        u8g2.drawButtonUTF8(64, 10, U8G2_BTN_HCENTER | U8G2_BTN_BW2 | U8G2_BTN_INV, u8g2.getDisplayWidth() - 2 * 2, 2, 0, "Menu"); 
        break;
    }
    return 0;
}

muif_t muif_list[] MUI_PROGMEM = {
    // MUIF_U8G2_LABEL(),
    MUIF_U8G2_FONT_STYLE(0, u8g2_font_helvR08_tr),
    MUIF_RO("DE", menu_static_part),
    MUIF_VARIABLE("EX", NULL, mui_u8g2_btn_exit_wm_fi)};

fds_t fds_data[] MUI_PROGMEM =
    MUI_FORM(1)
        MUI_AUX("DE")
          MUI_GOTO(64, 56, 2, "OK")
              MUI_FORM(2)
                  MUI_AUX("DE")
                      MUI_XYT("EX", 64, 56, " Exit ");

uint8_t is_redraw = 1;
`

But on my screen, there is no any "OK" button appeared in Form 1, so that I can not go to Form 2.

My setup: ESP32C3 with Arduino, SSD1306 Oled 128x64 I2C screen.
Could you please help me with this issue?

Thank you,

marhalloweenvt, 2024-11-15:
Thanks for your explanation. 
I have changed to use "Jump Button" right after I posted this issue and everything work flawlessly.
This is something out of scope of this issue, but I would like to ask if there is any a way to use "Number" for Float type and with specific incensement? Ex: Instead jump from 1.2 to 2.2, it can jump from 1.2 to 1.4 (+0.2 incensement).

olikraus, 2024-11-15:
I think it was discussed already somewhere here.

Maybe here:
https://github.com/olikraus/u8g2/issues/2316

And I think this could be a solution for your question: 
https://github.com/olikraus/u8g2/discussions/2499

I also think, this example should be part of the Arduino UI examples for u8g2 : https://github.com/olikraus/u8g2/blob/5a1779d0d095ea5e3eac196fe64c53114b29dba9/sys/arduino/u8g2_page_buffer/MUIFractionSimpleRotary/MUIFractionSimpleRotary.ino#L414-L418

marhalloweenvt, 2024-11-15:
Thank you. This is exactly what I mean.
I think my issue is completely resolved.
Thank for your hard work.

closed 2024-11-15 by marhalloweenvt

olikraus, 2024-11-15:
I will leave this issue open to remind myself to improve the MUI docs