olikraus/u8g2 #2538: MUI callbacks & setting parameter from list item - have I got this right?

clickworkorange, 2024-11-19:
First, many thanks for this impressive library - u8g2 on its own is just amazing with tons of functionality and a vast array of supported displays, an impossibly huge library of fonts, etc, etc - but that it *also* includes a fully functional GUI toolkit (MUI) just blows my mind. Browsing the code I feel like a Neanderthal looking at a Saturn V rocket... 

I'm relatively inexperienced with C (mainly on the 8-bit AVR platform), and total n00b with C++, but I'm stubborn and willing to learn :) I'm currently building a little gadget based on the equally mind-blowing Teensy 4.0, and have made good progress. It talks to another device over serial and I need to be able to configure the communication parameters (i.e. the Baud rate), for which I've built a simple menu using MUI. To my amazement this seems to work, but before I venture any deeper into the forest I'd like to clear up any potential misconception about how to best use the form element callbacks to modify parameters, specifically  with the `MUIF_U8G2_U16_LIST`. 

I understand this widget expects an array of strings as the list of elements to display, and that a callback function is executed when you change the selected item. This updates the index of the list, which is stored in a global `uint16_t`. Now I want to take that index and turn it into an *actual* Baud rate, and apply it to the serial port, when the user clicks on an "OK" button. So I've created a `struct` to hold the strings and their associated values, instead of a simple array of (pointers to) strings: 

```c++
uint16_t baud_rate_idx = 0;

struct BaudRate {
  const uint8_t idx; // not really needed
  const char* name;
  const uint value;
};

const BaudRate baud_rates[3] = {
  {0, "4800", 4800},
  {1, "9600", 9600},
  {2, "38400", 38400}
};
```

I have modified the default `get_cnt` and `get_str` functions to work with my `struct`: 
```c++
uint16_t baud_rates_get_cnt(void *data) {
  return sizeof(baud_rates)/sizeof(baud_rates[0]);
}
const char *baud_rates_get_str(void *data, uint16_t index) {
  return baud_rates[index].name;
}
```

Then I supply my own callback function for the button, which checks if `msg == MUIF_MSG_CURSOR_SELECT` (the button was clicked), and if so applies the chosen Baud rate to the serial port. Something like this: 

```c++
uint8_t mui_u8g2_btn_save_baud(mui_t *ui, uint8_t msg) {
  uint8_t result = mui_u8g2_btn_goto_wm_fi(ui, msg);
  if(msg == MUIF_MSG_CURSOR_SELECT) {
    mySerialSetBaudFunc(baud_rates[baud_rate_idx].value);
  }
  return result;
}

muif_t muif_list[] = {
  ...
  MUIF_U8G2_U16_LIST("NA", &baud_idx, NULL, baud_rates_get_str, baud_rates_get_cnt, mui_u8g2_u16_list_line_wa_mse_pi),
  MUIF_EXECUTE_ON_SELECT_BUTTON("GO", mui_u8g2_btn_save_baud),
  ...
}
```

My question is whether this is a good pattern to follow when using MUI? Have I perhaps misunderstood something fundamental?

olikraus, 2024-11-23:
I think there is no right and wrong. Actually I like your example. But let me do a small summary.
The question is: How to apply a user selection / modification to hardware?
I see the following options:
 * Extend a button function like you did in your approach
 * Use `MUIF_MSG_FORM_END` message as described in https://github.com/olikraus/u8g2/wiki/muimanual#stopwatch-example
 * Test for the user selection in the main event loop, similar to https://github.com/olikraus/u8g2/wiki/muimanual#muicountdown-example
 * You could introduce a second idx, for example `applied_baud_rate_idx`. In the event loop you check whether `baud_rate_idx` differs from `applied_baud_rate_idx`. If so, you copy the value of `baud_rate_idx` to `applied_baud_rate_idx` and do an update of the hardware register.

clickworkorange, 2024-11-23:
Thank you so much for the detailed response - I've actually carried on using the pattern I was asking about and it's great to hear I've not been building myself into a corner! Even better, the stopwatch example you mentioned made me realise how to implement _another_ thing I had been stumbling around with; how to display interactive non-MUI elements and widgets within a form. Thanks to reading that example through a few times, and with a bit of trial and error, I now have extremely fluid non-blocking live graphing inside a MUI form, which lets me use MUI elements and callbacks as part of the interface. Absolutely brilliant, love it!

closed 2024-11-23 by clickworkorange