olikraus/u8g2 #2545: ST7565R + STM32F103RCT6 SPI HW + CubeIDE Screen is blank

kQD1308, 2024-12-02:
[image]

I tested it on Arduino and it works well
`U8G2_ST7565_ERC12864_ALT_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ PB3, /* data=*/ PB5, /* cs=*/ PC12, /* dc=*/ PB6, /* reset=*/ PB7);`
But when I port u8g2 lib to CubeIDE + HAL, it does not work, just shows blank
Please help me fix this issue. Thank you !!!

**This is my code:**
```c
uint8_t u8g2_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{

    switch(msg)
    {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:
        break;

    case U8X8_MSG_DELAY_MILLI:
        HAL_Delay(arg_int);
        break;

    case U8X8_MSG_GPIO_CS:
        HAL_GPIO_WritePin(LCD_CS_GPIO_Port, LCD_CS_Pin, arg_int);
        break;

    case U8X8_MSG_GPIO_DC:
        HAL_GPIO_WritePin(LCD_DC_GPIO_Port, LCD_DC_Pin, arg_int);
        break;

    case U8X8_MSG_GPIO_RESET:
        HAL_GPIO_WritePin(LCD_RST_GPIO_Port, LCD_RST_Pin, arg_int);
        break;
    }
    return 1;
}

uint8_t u8x8_byte_stm32_hw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
    switch(msg) {
    case U8X8_MSG_BYTE_SEND:
        /* Insert codes to transmit data */
        if(HAL_SPI_Transmit(&hspi1, arg_ptr, arg_int, 1000) != HAL_OK) return 0;
            break;
    case U8X8_MSG_BYTE_INIT:
        /* Insert codes to begin SPI transmission */

        break;
    case U8X8_MSG_BYTE_SET_DC:
        /* Control DC pin, U8X8_MSG_GPIO_DC will be called */
        u8x8_gpio_SetDC(u8x8, arg_int);
        break;
    case U8X8_MSG_BYTE_START_TRANSFER:
        /* Select slave, U8X8_MSG_GPIO_CS will be called */
        u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);

        HAL_Delay(1);
        break;
    case U8X8_MSG_BYTE_END_TRANSFER:
        HAL_Delay(1);
        /* Insert codes to end SPI transmission */

        u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
        break;
    default:
        return 0;
    }
    return 1;
}

void lcd_run(void)
{
    u8g2_Setup_st7565_erc12864_alt_f(&u8g2, U8G2_R0, u8x8_byte_stm32_hw_spi, u8g2_gpio_and_delay_stm32);
    u8g2_InitDisplay(&u8g2);
    u8g2_SetPowerSave(&u8g2, 0);
    u8g2_ClearBuffer(&u8g2);
    u8g2_SetContrast(&u8g2, 160);

    while (1)
    {
        u8g2_FirstPage(&u8g2);
           do
           {
             u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
             u8g2_DrawStr(&u8g2, 0, 15, "Hello World!");
             u8g2_DrawCircle(&u8g2, 64, 40, 10, U8G2_DRAW_ALL);
           } while (u8g2_NextPage(&u8g2));
    }
}
```

**CubeMX Configuration:**

[image]

olikraus, 2024-12-02:
You should also support the other delay messages:
``` C++
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
      break;
```

Probably you also need to init the SPI subsystem. At least the corresponding code is not shown.

kQD1308, 2024-12-03:
I updated firmware package STMCubeF1 from v1.8.5 to v1.8.6 (latest). Now It works
I checked release notes and it seems that is related to my issue

[image]

closed 2024-12-04 by kQD1308