olikraus/u8g2 #2546: Add SSD1309 128x128 support / How to create u8g2 pull requests?

olikraus, 2024-12-06:
### Discussed in https://github.com/olikraus/u8g2/discussions/2543

<div type='discussions-op-text'>

<sup>Originally posted by **CSC3000** November 29, 2024</sup>
Hi Oli!
Ich habe eine kleine Erweiterung für ein Display SSD1309 128x128 hinzugefügt, kann aber keinen Branch mit meinen Änderungen erstellen.
Was ist der beste Weg um die Erweiterung in den Master zu bekommen?
Beste Grüße 
CSC

Hi Oli!
I have added a small extension for a Display SSD1309 128x128, but can't create a branch with my changes.
What is the best way to get the extension into the master?
Best regards 
CSC
</div>

olikraus, 2024-12-06:
PR: https://github.com/olikraus/u8g2/pull/2544

olikraus, 2024-12-06:
@CSC3000 Danke!

compiles without problem :-)

CSC3000, 2024-12-07:
Ebenfalls Danke!

closed 2026-05-17 by olikraus