olikraus/u8g2 #2553: Can the I2C bus be reused for other devices?

Xerxes31415, 2024-12-19:
I'm working on a project that uses this library to drive a display.  The new revision of the board added a power monitor (INA219), but I've been having a hell of a time integrating the two.  

The most common error I'm getting is `[E][Wire.cpp:526] write(): NULL TX buffer pointer`. I'm using the Adafruit INA219 Arduino library. 

Is there something fundamental to the u8g2 project that prevents sharing of the I2C bus?  
Is there an example project that uses the I2C bus for the both the display and another device?

Thanks in advance!

olikraus, 2024-12-19:
> Is there something fundamental to the u8g2 project that prevents sharing of the I2C bus?

Yes, the configuration of the I2C bus, see also note 2 here: https://github.com/olikraus/u8g2/wiki/u8g2reference#setbusclock

Assign the same clock rate used for the INA219 sensor also to u8g2.

Xerxes31415, 2024-12-19:
OK, good to know. I will give that a try. 
Thank you!