olikraus/u8g2 #2554: bmf2bdf.c file was detected as infected file from bitdefender

K-IF, 2024-12-19:
Hello,

After downloading the library in my computer my antivirus (BitDefender) detected the bmf2bdf.c file as an infected file.

I'm attaching the screenshots with the warning messages. 

[image]

[image]

Can you please also check this file with an antivirus or something also ? 

Can you please remove the file from the repository if this is infected ? 

Best regards,
kyf

olikraus, 2024-12-19:
The file 
https://github.com/olikraus/u8g2/commits/master/tools/font/bmf2bdf/bmf2bdf.c
is hand crafted source code. The history also doesn't show any unusual activity.
I assume your virus checker did issue a false positive warning.

olikraus, 2024-12-19:
To be on the safe side, I have enable code scanning now: https://docs.github.com/en/code-security/code-scanning/enabling-code-scanning/configuring-default-setup-for-code-scanning

K-IF, 2025-02-02:
Hello Olikraus,

I apologize for the delayed response. Thank you for your quick reply regarding this matter. I believe it must be a false warning from the antivirus, but it continues to flag this file as a threat. I’ve inspected the file in a text editor and did not find anything suspicious or any comments/lines of code that seem concerning to me.

closed 2025-02-02 by K-IF