olikraus/u8g2 #2557: wiki page u8g2setupcpp not complete

fablableo, 2024-12-23:
Hi olikraus,
I looked for the SSD1362 information on the u8g2setupcpp wiki page (https://github.com/olikraus/u8g2/wiki/u8g2setupcpp) but seemingly the page is cut off at a certain page. See attached screenshot:
[image]

Not sure what's happening. It stops halfway of the information on ssd1322-nhd_128x64. All other items below are just not there. I checked in chrome, firefox and edge. DevTools console contains no errors. Maybe there's a github wiki filesize limitation of some sort?

olikraus, 2024-12-24:
Thanks for the hint. 
This is the github copilot answer:

f your Markdown file appears to be truncated on GitHub, it could be due to a few reasons:

1. File Size Limit: GitHub's rendering only works for files up to 512KB. If your file exceeds this size, it might not render completely.
2. Complex Diffs: If you're viewing a diff of a Markdown file in a pull request or commit, and the changes are too complex or numerous, GitHub might not be able to render the entire diff. In such cases, you can still view the source view to analyze the changes.
3. Rendering Issues: If your Markdown file contains embedded HTML or other complex elements, some parts might not render as expected. It's a good idea to check both the rendered and source views for completeness.

So it looks liek there first answer is true in this case.

I guess I need to split the file :-/

olikraus, 2024-12-24:
top 5 md files:
```
kraus@acerb3:~/git/u8g2.wiki$ ls -alS *.md
-rw-rw-r-- 1 kraus kraus 558990 Dez 24 22:34 u8g2setupcpp.md
-rw-rw-r-- 1 kraus kraus 463368 Dez  6 22:32 u8g2setupc.md
-rw-rw-r-- 1 kraus kraus 136902 Dez  6 22:32 u8x8setupc.md
-rw-rw-r-- 1 kraus kraus 127987 Dez  6 22:32 u8x8setupcpp.md
-rw-rw-r-- 1 kraus kraus 108243 Sep 29 10:31 fntlistallplain.md
```

olikraus, 2024-12-24:
maybe I can als try to reduce the size a little bit

olikraus, 2025-04-18:
fixed for now...

closed 2025-04-18 by olikraus