olikraus/u8g2 #2559: SSD1322 5.5 OLED Display issue

fcig, 2024-12-30:
Hi, I'm having an issue with a SSD1322 OLED that I bought from the link below.
https://pt.aliexpress.com/item/1005006049846504.html

I'm using U8g2 through 4P SPI and I've tried all SSD1322 256x64 constructors, NHD, ZJY, sw spi, hardware spi, arduino nano, stm32 blue pill... no success at all, the display simply doesn't show anything. Looking for images and youtube videos about this display I've noticed that everyone is using the display with a different board with 2x8 pin header on one side with resistors to choose the comunication method, while mine has the SPI pins one one side and 8080 on the other with no resistors to choose.

Did anybody tested this specific board and have any suggestion that I could try? I think it's easier to say I'm doing something wrong than expect the board to be broken...

olikraus, 2025-01-07:
At least something should be visible.
The P8080 labeld SMD thing looks like a solder bridge. Did you remove this for your tests with SPI interface?

fcig, 2025-01-07:
Initially no, but after my post here I found another seller on aliexpress with one buyer comment saying to remove this bridge to work through SPI. I've tried that and still no success... I'm also confused about the specs because in some places there is information that this screen can work with 5v and in other places that it is 3v3 only, so I don't know if I damaged it doing this bunch of tests.

I've also bought a 3.1 OLED display, also with SSD1322 and the same 30 PIN connector behind... if I can make this new display work I'm willing to swap the boards to test the 5.5 screen, it should be compatible, right?

olikraus, 2025-01-07:
> some places there is information that this screen can work with 5v and in other places that it is 3v3 only

Be aware, that this not only applies to the power supploy, but also  to the GPIO connections. A 5V Arduino Uno may not be able to drive a 3.3V display. 

In the gallery https://github.com/olikraus/u8g2/wiki/gallery#2-oct-2016-ssd1322-256x64-nhd-display
there is an adapter board between the 5V Arduino Uno and the OLED. Not visible, but there is a 74HC4050 level shifter included to get 3.3V signals for the display. In Arduino world usually something like this https://www.az-delivery.de/en/products/logiklevel-wandler-3-3v-5v is used.

fcig, 2025-01-13:
Thank you very much for these tips. Considering what you wrote I probably damaged the display in my first test as I did it with arduino and 5v.

The 3.1" SSD1322 display arrived today and I was able to make it work using a blue pill (since I have no level shifters to use it with arduino) and u8g2 hello world.

I bought another 5.5" display and I will give another try. I will post here when it arrives for anyone else facing the same confusion as me.

fcig, 2025-01-20:
Got the new 5.5 display and it worked fine with your tips! Thanks!

Using 3.3v with a blue pill, P8080 SMD bridge removed to allow SPI, and the constructor below: 
`U8G2_SSD1322_NHD_256X64_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ PA4, /* dc=*/ PB0, /* reset=*/ PB1);  // Enable U8G2_16BIT in u8g2.h`

https://postimg.cc/QBhfgsgT

Yooky-Zooky, 2025-12-09:
> > Where would be the P8080 SMD bridge ?
> 
> There is a label "P8080" on the back of the screen board with a small "103" resistor below. It's in the center, easy to find. Just remove this resistor from the board.

Thanks a lot! after removing the 10KOhm in the middle, it was working! so simple and so badly documented from the store...
The screen is great when works ! ;-)