olikraus/u8g2 #2563: Request support for a large LCD with a resolution of 392x136 from Volkswagen

suwill, 2025-01-06:
Firstly, thank you to u8g2 for making DIY easier
I got a LCD from the Volkswagen radio, which has high resolution and large size.In China, this screen costs approximately 60 RMB, less than 10 euros, and includes a complete radio.
[image]
The definition of GPIO is as follows:
[image]
[image]

A netizen post  an example of C51(q395418298@http://bbs.mydigit.cn/read.php?tid=2537024):

[image]
[image]
[C51.zip](https://github.com/user-attachments/files/18312376/C51.zip)

I pay for another friend and added it to u8g2 2.27.6. It can be used normally,But now we have lost contact with him.
[U8g2.Addedtft3p3154@2020-08-08-v0.1.zip](https://github.com/user-attachments/files/18312390/U8g2.Addedtft3p3154%402020-08-08-v0.1.zip)

But now the version of u8g2 has been updated to 2.36.2,
I'm trying to figure out what he did:
[image]

I tried to copy some code from 2.27.6 to 2.36.2,Here are some operations I have performed:
```
//1. Copy the code below to U8g2\src\U8g2lib.h  line 282200
class U8G2_TFT3P3154_392x136_F_4W_SW_SPI : public U8G2 {
  public: U8G2_TFT3P3154_392x136_F_4W_SW_SPI(const u8g2_cb_t *rotation, uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) : U8G2() {
    u8g2_Setup_tft3p3154_392x136_noname_f(&u8g2, rotation, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
  }
};
```
```
//2. Copy the code below to \U8g2\src\U8x8lib.h line 9678
class U8X8_TFT3P3154_4W_HW_SPI : public U8X8 {
  public: U8X8_TFT3P3154_4W_HW_SPI(uint8_t clock, uint8_t data, uint8_t cs, uint8_t dc, uint8_t reset = U8X8_PIN_NONE) :U8X8() {
    u8x8_Setup(getU8x8(), u8x8_d_tft3p3154_392x136_noname, u8x8_cad_011, u8x8_byte_arduino_4wire_sw_spi, u8x8_gpio_and_delay_arduino);
    u8x8_SetPin_4Wire_SW_SPI(getU8x8(), clock, data, cs, dc, reset);
  }
};
```
```
//3. Copy the code below to U8g2\src\clib\u8g2.h line 1703
void u8g2_Setup_tft3p3154_392x136_noname_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb);
```
```
//4. Copy the code below to U8g2\src\clib\u8g2_d_setup.c line 9255
void u8g2_Setup_tft3p3154_392x136_noname_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_tft3p3154_392x136_noname, u8x8_cad_011, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_48_17_3(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);
}
```
```
//5. Copy the code below to U8g2\src\clib\u8g2_d_setup.c line 1056
uint8_t u8x8_d_tft3p3154_392x136_noname(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr);
```

```
//6. Copy u8x8_d_tft3P3154.c to U8g2\src\clib
// The following link
```
[u8x8_d_tft3P3154.zip](https://github.com/user-attachments/files/18312639/u8x8_d_tft3P3154.zip)

but vscode encountered an  error during compilation:

[image]
[image]
[image]
[image]
[image]

I am not familiar with the C/C++language, perhaps someone can tell me how to fix it and hope that the official can increase support for this screen.
I come from China and not very good at English. Sorry for using  translator software,Forgive my poor English.

suwill, 2025-01-06:
some photos
[image]
[image]
[image]

olikraus, 2025-01-07:
maybe I find some time to look into this in the future...

suwill, 2025-01-15:
> maybe I find some time to look into this in the future...

thanks for your attention

suwill, 2025-05-04:
any progress

olikraus, 2025-05-07:
didn't found time yet... too many other topics :-(