olikraus/u8g2 #2565: SSD1320 128x72 Display Constructor

Foxxify1, 2025-01-08:
Hello, I have a display that doesn't have a constructor yet. I got the constructor U8G2_SSD1320_160X80_1_4W_HW_SPI to work, but it appears to be off the screen a few pixels on top. I'm a newbie with coding so any help is appreciated!

The display I have is [this one](https://www.buydisplay.com/128x72-grayscale-oled-spi-white-0-72-inch-arduino-raspberry-pi)
the SSD1320's datasheet and the display's datasheet are attached on the website.

As a side question, why can't there be one constructor per controller IC (SSD1306, SSD1320, etc) and I pass in the resolution and other selections (1,2,F, HW_SPI, etc) with the pins to the constructor? Is it just the implementation, or is it something else?

olikraus, 2025-01-08:
> SSD1320's datasheet and the display's datasheet are attached on the website.

I have ordered one of those displays for my lab.

> s it just the implementation, or is it something else?

I try to keep it short.

## Static Display Memory

Assume, you would pass the resolution then there are two options to deal with this:
 * Dynamic memory allocation
 * Take the faction of a static preallocated memory block

Point one is risky, because the memory allocation might fail, especially in an embedded environment. It is usually better to get the out of memory warning at compile time (for example in Arduino IDE). To support this u8g2 had been designed in such a way that it doesn't use dynamic memory. 

Second point also doesn't make sense. Take your controller SSD1320: This is actually a 160x160, so u8g2 needs to reservere 160x160/8 = 3200 bytes (because the true size of your display is not known). Then later you pass 128x72 as a resolution, which would require only 1152 bytes. So 2/3 of the memory would be wasted.

So the u8g2 encodes the resolution into the constructor name, which then allocates the exact required amount of data at link time.

## Other Selections

The selection of the communication interface is thinkable, but ...
 * Each comm type has several flavors like 3-wire SPI or 4-wire SPI, but also differences in the SPI protocol level (which is even more worse for I2C). Also the number of pins depends. HW SPI often uses fixed assigned ports, but other uC are more flexible.
 * Actually the comm interface can be replaced:  https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform
 * Other selections are more difficult. A display is not specified by a single controller name like "SSD1306", it is instead a complete configuration bundle. The configuration includes:
     * Where is the LCD located within the controller memory: In your case: Where is the 128x72 area located within the 160x160 area of the controller? This needs to be specified. Unfortunately each controller provides different ways to specify this.
     * What are the electrical parameters required for the LCD? Different LCD capacity requires different changing configuration of the individual pixel (which in detail differs which each controller). 
     * What is the desired application? How fast should be the refresh rate of the display, what is the expected contrast? All this has to considered and is part of the configuration.
     * Changing one value always has an effect on other values. A different resolution always has an impact on the contrast and bias setup of the LCD and requires an re-adjustment of the same.

All in all the constructor name represents a specific display configuration. It may work for a display with the same resolution, but as soon as the resolution changes, we need to start from scratch...

Foxxify1, 2025-01-08:
Thanks for the in depth response!

I didn't know about the static display memory, but that makes total sense. Also I didn't realize there were so many differences between displays with the same driver ic, such as the position of the pixels within the controller's 160x160 area that needs to be taken in to account.

olikraus, 2025-02-09:
@Foxxify1 I have implemented support for u8g2. However the brightness is very low. Same is true for the Arduino Example from buydisplay.com
Do you have the same issue with the contrast? I wonder whether my display is somehow broken.

olikraus, 2025-02-09:
On the below image the contrast is actually better than in reality:
[image]

olikraus, 2025-02-09:
I have created beta 2.36.5 with the new constructor U8G2_SSD1320_128X72...

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

Foxxify1, 2025-02-24:
Sorry, I totally blanked on this, I've been busy learning linux for the first time in between long work days haha

> [@Foxxify1](https://github.com/Foxxify1) the brightness is very low.

Yes, the display spec is 100 nits. All of mine are also low, which is good for my use case in a dark room!

I'm going to test the lib later this week when I'm able to, thanks for your assistance!

closed 2025-02-24 by Foxxify1