olikraus/u8g2 #2568: How to count the chars of the string ?

eagl1, 2025-01-10:
Hi,

I've been trying to count the chars of a string, because I want to calculate the numbers digits I want to print.

For example, instead of printing number 1 in a 3 digit buffer size, when using:

`strcpy(buf, u8g2_u8toa(number, digit));`

So number 1 is printed like 001. So I want to pre-calculate the lenght of the string and count the chars inside the string.

So, it's only 1, then it's printed 1. If it's a number with 3 digits, for example, then it's printed as is.

I tried this method, but it's not consisted, and most the time doesn't give correct calculations.

```
U8G2_ST7920_128X64_1_HW_SPI u8g2(U8G2_R0, /* CS=*/ 10, /* reset=*/ 8);

void setup(void) {
  u8g2.begin();
  u8g2.setFont(u8g2_font_threepix_tr);
}

void loop(void) {
  const char *string_array = "41144987654";
  uint8_t string_width, char_width, chars_count;
  string_width = u8g2.getStrWidth(string_array);
  char_width = u8g2.getMaxCharWidth();
  chars_count = string_width/char_width; // the answer should be 11
  
  u8g2.firstPage();
  do {
    u8g2.drawStr(0,  10, "string_width");
    u8g2.drawStr(90, 10, convert_number(string_width, 3));

    u8g2.drawStr(0,  30, "char_width");
    u8g2.drawStr(90, 30, convert_number(char_width, 3));

    u8g2.drawStr(0,  50, "chars_count");
    u8g2.drawStr(90, 50, convert_number(chars_count, 3));
    
    
  } while ( u8g2.nextPage() );
}

const char *convert_number(uint16_t number, uint8_t digit) {
  static char buf[5];
  strcpy(buf, u8g2_u8toa(number, digit));
  return buf;
}
```

olikraus, 2025-01-10:
You could use the "strlen()" function.

eagl1, 2025-01-10:
No, actually, there's a mistake in my example.

It's not for strings, my goal is integers. So, anyway, I used this method and it worked.

```
const char *convert_number(uint16_t number) {
  static char buf[5];

  uint8_t digit_count;
  if((number/10) > 9){
    digit_count = 3;
  }
  else if(((number/10) < 10) && ((number/10) > 0)){
    digit_count = 2;
  }
  else if((number/10) < 1){
    digit_count = 1;
  }
  
  strcpy(buf, u8g2_u8toa(number, digit_count));
  return buf;
}
```

Then I call it with string function:

```
u8g2.drawStr(x, y, convert_number(number));
```

closed 2025-01-11 by eagl1