olikraus/u8g2 #2573: U8G2_RA8835_320X240_F_6800 can't use

rock599, 2025-01-16:
I use this U8G2_RA8835_320X240_F_6800 u8g2(U8G2_R0, 34, 35, 36, 37, 38, 39, 40, 41, /*enable=*/28, /*cs=*/30, /*dc=*/31, /*reset=*/33);

BUT ERROR

Insufficient memory; see https://support.arduino.cc/hc/en-us/articles/360013825179 for ways to reduce usage
The data area has exceeded the available space of the development board

Compilation error: The data area exceeds the available space of the development board

rock599, 2025-01-16:
I use arduino mega device

olikraus, 2025-01-16:
Arduino IDE? What is the exact error message?

rock599, 2025-01-16:
Yes,thank you for your reply.
 it' say: 
Insufficient memory; see https://support.arduino.cc/hc/en-us/articles/360013825179 for ways to reduce usage
Sketch uses 13262 bytes (5%) of program storage.
Global variables use 9997 bytes (122%) of dynamic memory, leaving -1805 bytes for local variables. Maximum 8192 bytes
The data area has exceeded the available space of the development board

Compilation error: The data area exceeds the available space of the development board

I think I'm going to use an older version, but I'm not sure which one to use

olikraus, 2025-01-16:
Ah, the RAM is too small to hold the complete graphics buffer of your display. You need to use the u8g2 page mode, where the display is split into smaller pages.
Use U8G2_RA8835_320X240_1_6800 ("1" instead of "F") along with first-page / next-page loop instead of sendBuffer().

rock599, 2025-01-16:
Ok, what I want to achieve is to display the pattern on RA8835 320X240, use read XBM and display, in fact I am not sure what the difference is with U8G2_RA8835_320X240_1_6800 or U8G2_RA8835_320X240_F_6800

olikraus, 2025-01-16:
The "1" constructor implements the "picture loop": https://github.com/olikraus/u8glib/wiki/tpictureloop
The main purpose is to save RAM, which is indeed required in your case.

In the arduino IDE look into the examples for HelloWorld.ino in full buffer and page puffer folders:

"F":
https://github.com/olikraus/u8g2/blob/5fab5b9da65496c9e96d1736449d9d5a73f1c8c4/sys/arduino/u8g2_full_buffer/HelloWorld/HelloWorld.ino#L404-L410

"1":
https://github.com/olikraus/u8g2/blob/5fab5b9da65496c9e96d1736449d9d5a73f1c8c4/sys/arduino/u8g2_page_buffer/HelloWorld/HelloWorld.ino#L444-L454

rock599, 2025-01-17:
```
#include <Arduino.h>
#include <U8g2lib.h>    //u8g2 2.20.13
#include "HG32240.h"

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>

U8G2_RA8835_320X240_1_8080 u8g2(U8G2_R0, 34, 35, 36, 37, 38, 39, 40, 41/*D0-D7*/, /*enable=*/28, /*cs=*/30, /*dc=*/31, /*reset=*/33);

#define u8g2_logo_320x240_width 320
#define u8g2_logo_320x240_height 240

void setup(void) {
  pinMode(29, OUTPUT); //RD HI
  digitalWrite(29, 1); 

  u8g2.begin(); 
}

void loop(void) {
  u8g2.firstPage(); 
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawXBMP(0,0, u8g2_logo_320x240_width, u8g2_logo_320x240_height,HG32240_2_bits);
  } while ( u8g2.nextPage() );
  
}
```

this's my code, It has been compiled, but there are still some errors. After uploading, the screen is all white, but the xbm data I gave it is not.

olikraus, 2025-01-17:
Did you place the picture in progmem area?

rock599, 2025-01-17:
Yes, I put it in another .h file and included it, but I don’t have the LCD now. Thank you for your help. In addition, I would like to ask: if the stn type lcd does not have a driver ic, can it be lit using arduino?

rock599, 2025-02-04:
It can be used, but it seems that if too much is displayed, garbled characters will appear.[image]

rock599, 2025-02-04:
I use Arduino mega and U8G2_RA8835_320X240_1_8080 u8g2.
Another photo is working[image]

olikraus, 2025-02-06:
nice

rock599, 2025-02-07:
But I don’t know why scan lines appear only in this picture. The video with delay 500ms is as follows

https://github.com/user-attachments/assets/5ce09ff8-62e2-4ac7-b49b-a80a0b4f43f0

olikraus, 2025-02-07:
I am confused, 3 days back you mentioned that it works.

rock599, 2025-02-07:
Sorry for the confusion. I tried six photos, and five of them were successfully displayed. Only this photo with a horizontal bar showed scan lines.

olikraus, 2025-02-07:
Hmmm not really clear how I can help here. Maybe you can post your full code along with a picture of what is wrong...

rock599, 2025-02-08:
```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define u8g2_logo_320x240_width 320
#define u8g2_logo_320x240_height 240
static const unsigned char HG3224_horizontal_bits[] U8X8_PROGMEM = {
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff,
[... 740 lines cut ...]
   0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
   0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00  };

U8G2_RA8835_320X240_2_8080 u8g2(U8G2_R0, 34, 35, 36, 37, 38, 39, 40, 41, /*enable=*/28, /*cs=*/30, /*dc=*/31, /*reset=*/33);  //picture1_u8g2

void setup(void) {
  pinMode(29, OUTPUT); //RD HI
  digitalWrite(29, 1); 

  u8g2.begin();  //picture1_u8g2
}

void loop(void) {
  u8g2.firstPage();        //55-59 load picture1(u8g2e)
  do {
    u8g2.setFont(u8g2_font_ncenB14_tr);
    u8g2.drawXBMP(0,0, u8g2_logo_320x240_width, u8g2_logo_320x240_height,HG3224_horizontal_bits);
  } while ( u8g2.nextPage() );
}

```
video is "VID_20250205_163351.mp4 "
The display is obviously darker. In other pictures, Vo=16.4V. In this picture, it needs 17.5V to display normally.

olikraus, 2025-02-09:
Maybe I missed something, but what exactly is the problem? To me all looks fine.

rock599, 2025-02-10:
[image]

[image]

I don’t know why the picture with only horizontal lines is unclear, but I didn’t change any voltage

rock599, 2025-02-10:
And when I increase the negative pressure, I will see something running

https://github.com/user-attachments/assets/b31bf634-21a0-470b-a21a-556e579ac69a

olikraus, 2025-02-12:
> I don’t know why the picture with only horizontal lines is unclear, but I didn’t change any voltage

Yes, I clearly see some lines. But isn't this what the code also should do?

I can only help if you provide a clear error description. Something like this:
Code looks like this:
u8g2.print("A");
But "B" appears on the screen.
Then I understand the problem.
If your code prints lines and your picture shows lines, then I don't see the problem.

rock599, 2025-02-13:
OK, thank you for your help.

closed 2025-02-14 by rock599