olikraus/u8g2 #2574: GP1294ai VFD Display freezes, not responding

loogcimon, 2025-01-17:
Hi, I have a simple running text with a counter. It works fine but after awhile like 30-40 minutes (the times are random) the display freezes and doesn't respond until I reset the ESP32. Could you help me with the issue?

`#include <U8g2lib.h> // u8g2 library is used to draw images on the display
#include <SPI.h>
//#include <Wire.h> // library required for IIC communication

U8G2_GP1294AI_256X48_F_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/7, /* dc=*/U8X8_PIN_NONE, /* reset=*/6);
//U8G2_GP1294AI_256X48_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/12, /* data=*/11, /* cs=*/7, /* dc=*/U8X8_PIN_NONE, /* reset=*/6);

int x, minX, x2, lineCount;
int dWidth=256;
String s="Successfully connected to the local network. All systems online.";

void setup() {
  Serial.begin(115200);
  u8g2.begin();
  u8g2.setContrast(3);
  u8g2.enableUTF8Print();
  u8g2.setFont(u8g2_font_chargen_92_tf);
  minX = -1 * u8g2.getStrWidth(s.c_str());
  x = dWidth;
  x2 = dWidth;
  lineCount=1;
}
int tme=0;
void loop() {
  u8g2.clearBuffer();
  u8g2.setCursor(x,25);
  u8g2.print(s);
  x--;
  if(x < (minX+200)){
    u8g2.setCursor(x2,25);
    u8g2.print(s);
    x2--;
    if(x < minX){
      x = x2;
      x2 = dWidth;
      lineCount++;
      Serial.println(lineCount);
    }
  }
  u8g2.setCursor(40,43);
  u8g2.print(lineCount);
  u8g2.sendBuffer();
  delay(20);
}`

[image]

I've tried software SPI it's working slow but with no freezes and seems like the issue occurs only in hardware SPI mode.

olikraus, 2025-01-17:
My knowledge on the ESP32 is very limited, but I heard from users, that the watchdog sometimes as an issue. Solution could be to disable watchdog timer: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/hardware-abstraction.html#hw-abstraction-hal-layer-disable-rtc-wdt

Not at all sure, whether this is a solution here...

loogcimon, 2025-01-18:
> My knowledge on the ESP32 is very limited, but I heard from users, that the watchdog sometimes is an issue. Solution could be to disable watchdog timer: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/hardware-abstraction.html#hw-abstraction-hal-layer-disable-rtc-wdt
> 
> Not at all sure, whether this is a solution here...

Thanks, will try