olikraus/u8g2 #2575: For ST7571, 128x96 it does not work correctly

jrobrtsn, 2025-01-19:
The top ~⅓ of the display is unable to be used for the unit I have. When using the 128x128 constructor it fills this part of the display but the whole display is very faint. [image]

[image]

Thanks for any help/fixes.

olikraus, 2025-01-19:
hmm... you can change the contrast manually by using u8g2.setContrast()

Just for my clarification: What kind of display do we see here (resolution, data sheet, link)?
What is the constructor name for both of the pictures?

jrobrtsn, 2025-01-19:
Hello, the display is a 'ccsb4736w' it is 128*96 resolution. It has 2-bit grey scale.
I've looked for data sheets but there is almost nothing. There's 3 results on Google and these just tell me the pin out of the display and that is it. So no datasheet to speak of. 

For the case of the top third being blank: 
`U8G2_ST7571_128X96_1_4W_HW_SPI`
for the case of the display being very faint:
`U8G2_ST7571_128X128_1_4W_HW_SPI`
I have tried the contrast setting. Set it to 255 and it makes the display much less faint for the 128X128 constructor. 
it makes the bottom ~⅔ of the screen completely black for the 128X96 constructor. 

The display is usable enough for what I plan to do with it, and with the lack of a datasheet that might be as good as it gets. 

Thanks.

olikraus, 2025-01-19:
Thanks, indeed the most usefull info seems to be this: https://github.com/SocialSisterYi/Light-Your-Screen/tree/main/CCSB4736W
So, lets call this display G12896 (because it is called like this in the above github page)

As a summary: 
The lower image is the U8G2_ST7571_128X96_1_4W_HW_SPI constructor: Contrast ok, but display area is shifted
The upper image is the U8G2_ST7571_128X128_1_4W_HW_SPI constructor: Contrast is wrong, dimension doesn't fit to your display.

Can you do a test for me: Please use the U8G2_ST7571_128X96_1_4W_HW_SPI and make a call to sendF:
``` C++
u8g2.sendF("ca", 0x44, 0x20);
```
after u8g2.begin(). Will this fix the 1/3 downshift of the content?

jrobrtsn, 2025-01-19:
thanks, the sendF call seems to fix it entirely. 

[image]

olikraus, 2025-01-19:
Great...

I have created a new beta release 2.36.3 with the new constructor "U8G2_ST7571_G12896_1_4W_HW_SPI".
It would be great if you could crosscheck, whether this constructor works.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
Arduino IDE:
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

PlatformIO:
`platformio.ini` (https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) should include 
```
lib_deps =
  u8g2=https://github.com/olikraus/U8g2_Arduino/archive/master.zip
```

jrobrtsn, 2025-01-19:
yes that new constructor works perfectly. thanks.

olikraus, 2025-01-19:
Nice, thanks for testing

closed 2025-01-21 by jrobrtsn