olikraus/u8g2 #2576: how to display NT7086 320X240

rock599, 2025-01-20:
The pinout diagram going to the LCD looks like this:

1 D0 Data Pin
2   D1 Data Pin
3   D2 Data Pin
4   D3 Data Pin
5   DISPOFF Controls Display off, 0: off, 1: on
6   FLM First Line Mark / Shift Direction Select
7   M Alternation Control Signal
8   LP, data latch (row) / Display Data I/P Clock
9   CP, data shift (column) / Display Data I/P Clock
10   VDD+5V Power Supply 5volts
11   VSS-GRD Ground
12   VLC-24V
13   VO Contrast Adjustment, usually uses POT that can source 0-5V
14   FGND

olikraus, 2025-01-20:
Not sure about the question here, but a link to the display itself would be helpful.

rock599, 2025-01-22:
At present, I have STM32F401 and ARDUINO MEGA, I am not sure whether it can be achieved.