olikraus/u8g2 #2578: getStrWidth() returns wrong estimation if "°" in input String

MaBaTa, 2025-01-20:
I have been tring to display a temperature horizontaly centered with the use of getStrWidth().

But the String is sligtly of to the left, so i assume the estimation of getStrWidth() counts the "°" as a wider symbol than it is.

Yes i have checked without the "°" symbol and with different fonts, it always worked without and never with the Symbol.

Happy if i can help to improve this absolutly great library.
Thanks!
MaBaTa

olikraus, 2025-01-21:
Thanks for your feedback, however this is not a bug. Instead your string "°" probably contains an UTF-8 encoded glyph, which can't be handled by getStrWidth(), instead use [getUTF8Width()](https://github.com/olikraus/u8g2/wiki/u8g2reference#getutf8width). 
I have updated the documentation for [getStrWidth()](https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth) by adding a warning.

MaBaTa, 2025-01-21:
Thanks for your fast response!

after switching to [getUTF8Width()](https://github.com/olikraus/u8g2/wiki/u8g2reference#getutf8width) everything works like expected, Thanks again!

with "°" i ment `'°'` so the degree symbol, i thought that isn't a special glyph.
And after checking it actually is part of the extended ASCII Table, so [getStrWidth()](https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth) works ony for the non extended one?

MaBaTa, 2025-01-21:
> I have updated the documentation for [getStrWidth()](https://github.com/olikraus/u8g2/wiki/u8g2reference#getstrwidth) by adding a warning.

*Little typo there by the way*

olikraus, 2025-01-22:
> for the non extended one?

It has nothing todo with u8g2 or u8g2 fonts. The simple truth is: You have entered an UTF8 string, so you have to use the UTF8 U8g2 functions.

You might not know, but almost all editors (including Arduino IDE) will use UTF8.
Actually your string contains three bytes: 0xC2 0xB0 0x00 (the last byte is the string terminator)
The byte sequence 0xC2 0xB0 is called UTF-8 sequence and describes the degree symbol.

And yes, at the end you need a font which includes the degree symbol, otherwise the width will be unknown.