olikraus/u8g2 #2581: SH1107 128x128 1.5" I2C display offset

tomastvrdy0, 2025-01-28:
Arduino UNO with this display https://www.laskakit.cz/en/oled-displej-bily-sh1107-128x128-1-5--i2c/ running the u8x8 helloworld example sketch.

U8G2_SH1107_128X128_1_HW_I2C u8g2(U8G2_R0);

Display very offset

Any way to change the offset?

[image]

olikraus, 2025-01-29:
You could try 

```  C++
u8g2.getU8x8()->display_info->default_x_offset = 0;
```

before u8g2.begin().
Will this work?

tomastvrdy0, 2025-01-29:
> You could try
> 
> u8g2.getU8x8()->display_info->default_x_offset = 0;
> before u8g2.begin(). Will this work?

It doesn´t work unfortunately, It shows this error:
exit status 1

Compilation error: assignment of member 'u8x8_display_info_struct::default_x_offset' in read-only object

olikraus, 2025-01-30:
oh, ok. Can you located the following file and line on your local harddisk?
https://github.com/olikraus/u8g2/blob/ef893381f9385df6e3351483272c31ccb89172e3/csrc/u8x8_d_sh1107.c#L519

Please change the value to 0. Will this fix the problem?
If so, I can create a new constructor.

tomastvrdy0, 2025-01-31:
> oh, ok. Can you located the following file and line on your local harddisk?
> 
> [u8g2/csrc/u8x8_d_sh1107.c](https://github.com/olikraus/u8g2/blob/ef893381f9385df6e3351483272c31ccb89172e3/csrc/u8x8_d_sh1107.c#L519)
> 
> Line 519 in [ef89338](/olikraus/u8g2/commit/ef893381f9385df6e3351483272c31ccb89172e3)
> 
>  /* default_x_offset = */ 96, 
> Please change the value to 0. Will this fix the problem? If so, I can create a new constructor.

It fixed it.
Thank you so much :))

closed 2025-01-31 by tomastvrdy0