olikraus/u8g2 #2585: SSD1306 - character pixels in left-most two columns missing when using drawStr

bmoerdyk2, 2025-02-05:
I have a ESP32 Dev Module with SSD1306 based display that I have used to create a clock.
The spacing and size of the characters is critical.
I got it always perfect, but the left-most two columns of pixels are not being rendered (e.g. X=0, X=1) when I use drawStr to show the numbers.  I even added a line to the display from (0,46)-(127,46) to prove the edges of the display.

I've attached the best picture I can get of the worst example of the pixels not being rendered.  The tens digit in the hour position is rendered AFTER the tens and ones digits of the minutes - so it's not a case of the character glyph from the minutes ones position wrapping around in memory to the left side of the display (something I've seen, and know to work aorund)

[image]

bmoerdyk2, 2025-02-05:
Using version 2.35.30 of the u8g2 library.  Two fonts at least show this problem: u8g2_font_logisoso46_tn and _fur42_tr
I have other projects using smaller fonts, and the text renders even into X=0 column. (e.g. u8g2_font_8x13_tr)

Some code, since the question will likely come up...  clockMinT, clockMinO, clockHourT and clockHourO are char arrays that I've udes sprintf to fill with the character for the respective digit.
`  u8g2.setFont(u8g2_font_fur42_tr);
  u8g2.setFontDirection(0);
    tw = int(u8g2.getUTF8Width(clockMinT));
    u8g2.drawStr(MinXT-tw/2, Row1Y, clockMinT);
    tw = int(u8g2.getUTF8Width(clockMinO));
    u8g2.drawStr(MinXO-tw/2, Row1Y, clockMinO);
    //hours
    tw = int(u8g2.getUTF8Width(clockHourT));
    u8g2.drawStr(HourXT-tw/2, Row1Y, clockHourT);
    tw = int(u8g2.getUTF8Width(clockHourO));
    u8g2.drawStr(HourXO-tw/2, Row1Y, clockHourO);
`

olikraus, 2025-02-06:
I am not able to reproduce the problem...
https://github.com/olikraus/u8g2/blob/b1db4b22b5b30c0a7edb47058459bbba037fc152/sys/sdl/issue2585/main.c#L49-L71

This is the generated image:
[image]

bmoerdyk2, 2025-02-06:
Thank you very much for taking the time to research this.  From your image, if I understand the colors... you still have 2 columns of unused pixels to the left of the left-most 0.  And... I realize I didn't give you values for the X-coordinate defines.  Based on the image, and my assumption about the free pixels, could you change HourXT to 13 and retry?  If I'm right, this should put the left-most edge of the 0 into column 0.  I think a value of 11 would result in the 0 being cut off because of the edge of the screen

[image]

olikraus, 2025-02-06:
>  I realize I didn't give you values for the X-coordinate defines. 

True, I just guessed some values. Indeed many questions are open. Maybe you could send little bit more of your code...

>  could you change HourXT to 13 and retry?

Done:

[image]

I also set the pixel at 0,0 for reference

> From the code you posted, it also looks like you're taking a screenshot from the microcontroller memory?

No, it is the SDL port of the the original code. Although I use C, it is exactly the same code like on any uC.

bmoerdyk2, 2025-02-07:
Thanks again.  I really appreciate your support.

I've uninstalled the library and reinstalled, and went back to the basic "HelloWorld" example file.
Arduino IDE 2.3.4.  U8G2 Library 2.34.30, freshly unisntalled and reinstalled.
"ESP32 Dev Module" as the board type

I've just modified "HelloWorld.ino", so here is the entire code listing with the unnecessary constructor lines deleted

`#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

// U8g2 Contructor List (Frame Buffer)
U8G2_SSD1309_128X64_NONAME2_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 23);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_fur42_tr);	// choose a suitable font
  u8g2.drawStr(-4,45,"00");	// write something to the internal memory
  u8g2.drawLine(0,22,127,22);
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);  
}`

[image]

bmoerdyk2, 2025-02-07:
Hi!  I found a work-around!!

  u8g2.drawStr(123,45,"0");	// write something to the internal memory     <-------------- new
  u8g2.drawStr(-5,45,"00");	// write something to the internal memory   <-- changed to -5 to align with edge of display

[image]

olikraus, 2025-02-07:
Maybe the internal x-offset is different with your display. Can you run the FlipMode exampe?
The frame should exactly fit into the border of the display. If this is not the case, then we need to adjust the x-offset.

I am also a confused: In your code you have used SSD1309, but issue title is SSD1306.
Maybe your display is a sh1106. The sh1106 differs in two pixel offset from SSD1306.

bmoerdyk2, 2025-02-07:
Hi, sorry... I'm not usually that sloppy.  As far as I know...it is SSD1309.  This is at least what was sold.   The seller's description of the product on Amazon claims it is SSD1309 based, but it likely came from China, so who knows.

I see why you asked to run FlipMode...the outline around the edge of the screen, or what should be the edge of the screen.
I am seeing a consistent offset of 2 pixels in the x-direction.
To the right with F0, to the left with F1

as an experiment, I changed to U8G2_SH1106_128X64_NONAME_1_HW_I2C u8g2(U8G2_R0, /* reset=*/ 23); 
Still getting the same offsets I described above.

I can keep on with my work-around, or how would I change the offset?  I did a quick windows search of the u8g2 library folder, it doesn't seem the text SSD1309 shows anywhere.

I found the offset setting in the u8x8_d_ssd1309.c file and have adjusted it to 0.
Compiled, downloaded, all good.  Box fits perfectly on the display with F0.
`  /* default_x_offset = */ 0,
`

bmoerdyk2, 2025-02-08:
Hallo Oli,
Es scheint, das Problem ist komplett gelöst.
vielen vielen Dank für die Hilfe!!
Grüße aus Berlin,
Brian

closed 2025-02-08 by bmoerdyk2

olikraus, 2025-02-09:
Na dann... :-)